bigtreetech/CB1 #93: Logitech C920 USB Camera Not working

matslade, 2023-02-18:
I am not able to get my C920 camera working the CB1 does see if when it is plugged in.

JaseHeuer, 2023-03-02:
This is the same issue that I have run into. I am running octoprint on the CB1+Pi4B combo.

medicusdkfz, 2023-03-22:
My C920 is running with a C1 - temorarely… I guess there is an issue with the current load of the USB ports.

mrstone14, 2023-10-21:
Is anyone looking into this issue? I'm having the same problem and it's causing "Lost communication with MCU" errors during prints. Would love to be able to use my camera without having to use a separate pi or workaround solution.

Pneumanifest, 2023-10-21:
It's not completely unique to the CB1 as I have a raspberry pi 3b+ that I get no signal with some cameras too. I think it's a Crowsnest glitch somewhere, because the same camera works fine when running octoptint.