bigtreetech/CB1 #94: Wireless connection not working on minimal image 2.3.1

VIXIVIXIV, 2023-02-18:
Since minimal image 2.2.1, connecting to wifi has been extremely unreliable. On initially booting, the board often won't list all networks that could be connected to using image 2.2.0, and the networks that do show up cannot be connected to. Adding the desired network to the connection script does not help either, NetworkManager will attempt to connect and list the connection as active, but no data can be transferred. If NetworkManager is restarted, the wifi breaks completely and doesn't even show as an option in NetworkManager, let alone show networks to connect to. Ethernet does work on these images. Only once, on my first attempt at using 2.3.1, was I able to connect to wifi, but the image bricked due to other causes, and on further attempts to install 2.3.1 using the same SD card it did not work. Maybe images since 2.2.1 have been tuned for a mid-lifecycle SKU change?

VIXIVIXIV, 2023-04-03:
Quick and overdue upate: version 2.3.2 does not solve this issue, maybe it's a step on the way to a fix but wireless still is broken

avseredyuk, 2024-03-21:
Any updates on the issue? Having similar problem with v2.3.4