bigtreetech/CB1 #97: CB1 + Manta M4P doesn't output HDMI

kieraneglin, 2023-02-26:
I have [this](https://a.co/d/bHlh0FM) screen which I know works and I have previously used with KlipperScreen. When I plug it into the Manta, it powers up but says there's no HDMI signal. I've tried several different cables and I've confirmed that the screen works with a Raspberry Pi 4. 

Is there some sort of configuration change needed to enable HDMI?

frutabruta, 2023-03-01:
I have very similar screen and it does not work with CB1 either, even though it works with RPi4. I ended up using 5 inch HDMI display from BTT.

Will-wastelander, 2023-03-01:
@frutabruta Did you follow in instructions in the manual ? You have to
download a file and put it in /boot.

kieraneglin, 2023-03-01:
@Will-wastelander can you link to these instructions? The readme talks about editing `/boot/BoardEnv.txt` but that did nothing in my case.

Will-wastelander, 2023-03-01:
@kieran Sorry, I mixed up my issues. For some reason I thought this was for
CM4 and dsi display.

kieraneglin, 2023-03-01:
@frutabruta after doing a LOT of digging, it seems like it's a hardware issue. In my case, I _think_ it's my V2.1 Manta M4P that's the issue and a V2.2 would solve things. It may be the CB1 revision version as well, but I can't seem to find any revision numbers on my board to compare.

Will-wastelander, 2023-03-01:
Version on CB1 should be right next to the CPU.

Will-wastelander, 2023-03-01:
[image]

frutabruta, 2023-03-01:
> @frutabruta after doing a LOT of digging, it seems like it's a hardware issue. In my case, I _think_ it's my V2.1 Manta M4P that's the issue and a V2.2 would solve things. It may be the CB1 revision version as well, but I can't seem to find any revision numbers on my board to compare.

You might be right about the CB1 version. I've just tried it in my CB1 V2.2 + Manta M8P and it seems to work. I think previously I tried CB1 V2.1 with Pi4B adapter.
[image]

kieraneglin, 2023-03-01:
I've confirmed that both of my CB1s are V2.2 so the only difference for me is the mainboard. V2.2 works,V2.1 does not. My hunch is that something in the general V2.1 CB1 IO configuration is the issue. @frutabruta if you still have the Pi4B adapter, would you mind checking if there's a version number on it?

kieraneglin, 2023-03-01:
Very interestingly, my CB1s run consistently 8-10º C warmer in the V2.1 M4P vs. the V2.2. I'm not going to dig any deeper into that, but I thought it was worth bringing up

frutabruta, 2023-03-01:
@kieraneglin the Pi4B adapter is version 1.0 and I don't think there have been any newer revisions.