bigtreetech/CB1 #99: linux-u-boot-h616-current package has missing 'Description' field

kmarty, 2023-03-08:
Which makes messages like this:
```
dpkg: warning: parsing file '/var/lib/dpkg/status' near line 8930 package 'linux-u-boot-h616-current':
 missing 'Description' field
```
anytime when system working with packaging system.

adjustinthings, 2023-04-28:
Yes I am seeing this now as well. Having trouble getting mosquitto broker installed as well if maybe thats something you might be doing as well.

zazzel23, 2023-07-30:
I've been seeing this, too. I believe it started the moment I installed the adxl345. Any solution yet?

Pneumanifest, 2023-08-25:
This is just a warning. It's saying that the discription field os the Linux-u-boot-h616 package is empty. It doesn't effect any functionally. Update to the newest cb1 image where they added a discription so the warning stops showing.

closed 2024-03-08 by bigtreetech