bigtreetech/CB1 #100: The default username and password are fixed

idlesilver, 2023-03-10:
The username and password cannot been modified with "raspberry-pi-imager"
It is fixed to be:
username = biqu
password = biqu

It would be nice to highlight the limitation explicitly!

z-shaffer, 2023-09-08:
If I change the name of Biqu like that, Mainsail stops working, it won't connect to Moonraker. Maybe it's because I had to run it like this multiple times quickly to get it to work, since biqu always has processes running:

`sudo killall -u biqu && sudo usermod -l newname biqu`

But a reboot doesn't fix it. Is there no way to configure the username or password before using the cb1 image?

MacBandito, 2023-12-05:
This is actually a problem.  Even when you SSH into the CB1 and use the sudo passwd command and change the password it doesn't change.  It says it does but on reboot it's still the default biqu.

drewgwallace, 2024-02-12:
> This is actually a problem. Even when you SSH into the CB1 and use the sudo passwd command and change the password it doesn't change. It says it does but on reboot it's still the default biqu.

To change the password of the biqu user, don't use sudo. 

`biqu@xolotl:~$ passwd
Changing password for biqu.
Current password:
New password:
Retype new password:
passwd: password updated successfully`

Worked for me on reboot.