bigtreetech/CB1 #101: ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32

Dids, 2023-03-12:
Long story short: I'm getting frequent(ish) but random USB disconnects while printing with CB1 + Klipper.

Are there known issues around the USB to serial hardware or kernel driver?
Or could this be caused by something else entirely, and if so, what exactly?

My prints get randomly stopped when Klipper disconnects and just leaves everything running.
However, I found the following in `dmesg`, which may prove useful:

```dmesg
[ 1792.333387] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[ 1792.365028] ch341-uart ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[ 1792.369401] ch341-uart ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[ 1792.373966] ch341-uart ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[ 1792.378273] ch341-uart ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[ 1792.382386] usb 2-1-port2: disabled by hub (EMI?), re-enabling...
[ 1792.382425] usb 2-1.2: USB disconnect, device number 3
[ 1792.382909] usb 2-1.2: ch341_read_int_callback - usb_submit_urb failed: -19
[ 1792.383588] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[ 1792.383674] ch341 2-1.2:1.0: device disconnected
[ 1792.727203] usb 2-1.2: new full-speed USB device number 4 using ehci-platform
[ 1792.940276] usb 2-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 1792.940308] usb 2-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1792.940317] usb 2-1.2: Product: USB Serial
[ 1792.941092] ch341 2-1.2:1.0: ch341-uart converter detected
[ 1792.943021] usb 2-1.2: ch341-uart converter now attached to ttyUSB1
[ 1792.943540] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[ 1966.718649] usb 2-1.2: USB disconnect, device number 4
[ 1966.719209] ch341-uart ttyUSB1: ch341-uart converter now disconnected from ttyUSB1
[ 1966.719279] ch341 2-1.2:1.0: device disconnected
[ 1971.351980] usb 2-1.2: new full-speed USB device number 5 using ehci-platform
[ 1971.581214] usb 2-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 1971.581245] usb 2-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 1971.581255] usb 2-1.2: Product: USB Serial
[ 1971.582028] ch341 2-1.2:1.0: ch341-uart converter detected
[ 1971.583260] usb 2-1.2: ch341-uart converter now attached to ttyUSB1
[ 1971.584366] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[ 1983.600963] usb 2-1.2: USB disconnect, device number 5
[ 1983.601533] ch341-uart ttyUSB1: ch341-uart converter now disconnected from ttyUSB1
[ 1983.601612] ch341 2-1.2:1.0: device disconnected
[ 2000.560123] usb 2-1.2: new full-speed USB device number 6 using ehci-platform
[ 2000.790010] usb 2-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 2000.790041] usb 2-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 2000.790051] usb 2-1.2: Product: USB Serial
[ 2000.792643] ch341 2-1.2:1.0: ch341-uart converter detected
[ 2000.794619] usb 2-1.2: ch341-uart converter now attached to ttyUSB1
[ 2000.795116] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
```

arthur12341994, 2023-11-18:
Hi been getting the similar issue,

Here is my `dmesg` output:

`[Fri Nov 17 15:14:46 2023] usb 2-1.4: reset high-speed USB device number 5 using ehci-platform
[Sat Nov 18 05:18:45 2023] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[Sat Nov 18 05:18:45 2023] ch341-uart ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[Sat Nov 18 05:18:46 2023] usb 2-1.2: USB disconnect, device number 3
[Sat Nov 18 05:18:46 2023] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[Sat Nov 18 05:18:46 2023] ch341 2-1.2:1.0: device disconnected
[Sat Nov 18 05:18:48 2023] usb 2-1.2: new full-speed USB device number 6 using ehci-platform
[Sat Nov 18 05:18:48 2023] usb 2-1.2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice=81.34
[Sat Nov 18 05:18:48 2023] usb 2-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[Sat Nov 18 05:18:48 2023] usb 2-1.2: Product: USB Serial
[Sat Nov 18 05:18:48 2023] ch341 2-1.2:1.0: ch341-uart converter detected
[Sat Nov 18 05:18:48 2023] usb 2-1.2: ch341-uart converter now attached to ttyUSB1

Here is my `dmesg | grep usb` outpu, it seems that all the usb connection were reset when the printer connection disconnected, but cannot find anything that may be causing this. Any suggestions?
`
These disconnects seemingly happen randomly for me at least, not entirely sure what is causing this.

bittz128, 2023-12-07:
My issue was that the buck converter voltage dropped just enough to cause
power issues.  Upped the output and we're back in business.