bigtreetech/CB1 #104: Connect temperature sensor?

daTobi1, 2023-03-19:
Hi, 

is it possible to connect a 1 wire temperature sensor on the cb1 with rpi adapter plate on the gipo pins?

TristanScott44, 2023-06-06:
I've been trying to figure out how to enable 1-wire on the CB1 as I would like to use a DS18B20 as my chamber temperature sensor.  Sounds like a lot of other people were having trouble with it too.  If I figure it out, I'll report back.

Puzzl88, 2023-09-27:
> I've been trying to figure out how to enable 1-wire on the CB1 as I would like to use a DS18B20 as my chamber temperature sensor. Sounds like a lot of other people were having trouble with it too. If I figure it out, I'll report back.

Were you able to connect the sensor?

mdvorak, 2023-09-29:
Please see https://github.com/bigtreetech/CB1/issues/65 for the answer.