bigtreetech/CB1 #106: Cb1 with Obico

Vinnnocc, 2023-04-04:
Anyone running a CB1 with Obico?
I noticed I’m getting still frame video every few seconds vs a steady fps or even buffering 

will CB1 ever be ready for obico?

Pneumanifest, 2023-04-04:
What resolution do you have set, and do you have the premium or basic version?

Vinnnocc, 2023-04-04:
> 

640x480 And premium.   I know when I installed mainsailOS on a pi version 64bit did 0.1FPS. Not this 10sec still frame  So I switched to 32bit mainsailOS and had full 30fps easy.    So not sure if cb1 can switch to 32bit or if it’s just not supported

Pneumanifest, 2023-04-04:
I'm not going to say it's not something to do with the way the OS is set up. I did install Obico to test and yeah it says it failed to stream and set it to 0.1fps, but what is weird is that OctoEverywhere streams fine. So it makes me think it's looking for something that's not installed, or there's some sort of odd configuration it wants.

Will-wastelander, 2023-04-04:
> I know obico is setup in code to look for Pi. So maybe it’s something on
> their end to have the code look for any kinda of non- PI to be able to
> stream

Obico does not work on 64bit OS. If you are running a 64bit OS, you will
need to switch to a 32bit OS and test again. If you are running a 32bit OS,
you can open a ticket with obico and confirm if the board is supported or
not.

Vinnnocc, 2023-04-04:
> 

Can I pick or choose what OS is installed? I thought when I installed everything on Cb1 everything was auto

Will-wastelander, 2023-04-04:
The image you use is what determines 32 or 64 bit.

Vinnnocc, 2023-04-04:
> The image you use is what determines 32 or 64 bit.
> […](#)

It was a cb1 image off here   With klipper in it nothing about mainsail 32-64

Vinnnocc, 2023-04-04:
[image]

Vinnnocc, 2023-04-04:
> It's a 64bit setup.
> Looking in the obico log there are some weird things. The first is that there is a warning that it's not on a pi and the stop streaming. The next is an error with syntax in "backoff" waiting for Janus and a connection refused error. Then after so many tries it just quits.

Obico is looking for a Pi and a will only stream on a Pi.   So not sure if cb1 can trick obico or we gotta wait for obico to support non-pi

Pneumanifest, 2023-04-04:
It says it's closing the video_pipeline that might be a Raspbian specific method, there is a feed trying to get through I can now see about 3 frames before it goes to buffering and I no longer get the 0.1fps error. Turns out I was missing libnss3-dev. After installing that, the backoff connection refused error went away.

Vinnnocc, 2023-04-04:
I can see video whole time but it gives me a signal frame every like 10sec.  Ai works still

Pneumanifest, 2023-04-05:
>   Ai works still 

Yeah, not having the stream working does not effect the rest of the features.

Vinnnocc, 2023-04-05:
I see now octoeverywhere is now fully supported by klipper might have to give that a try

Pneumanifest, 2023-04-05:
That's what I'm using on my machines. OctoEverywhere seems to be working fine. And it's a direct connection to the UI. You can also use it other apps like mobileraker

Vinnnocc, 2023-04-05:
Well that’s it I’m done using obico. Octoeverywhere installed supper easy and I’m reviving over 50FPS video feed

Pneumanifest, 2023-04-05:
Hahaha yeah it works great.

Pneumanifest, 2023-04-06:
So it turns out Obico indeed will only stream with raspberry pi as per design.
 https://obico.io/docs/user-guides/webcam-stream-stuck-at-1-10-fps/#2-is-octoprint-running-on-a-raspberry-pi

 I'm sure they have their reasons, but with all the other boards selling now, they are really limiting their market.

illspirit, 2024-03-14:
Man this is upsetting, CB1 should be supported.

Pneumanifest, 2024-03-15:
> Man this is upsetting, CB1 should be supported.

It's not the CB1 that doesn't support it, it's Obico not supporting video for non raspberry pi devices. Try out OctoEverywhere you can fully access your machine from anywhere and it streams video fine from the CB1. It also has an AI failure detection.