bigtreetech/CB1 #110: CB1 eMMc not booting up

Suggy50, 2023-04-20:
Manta M5P with CB1 eMMc.
Followed the instruction all the way upto flashing the internal memory with the Klipper image from your github but it wont boot so I can get an ip address. same if I use ethernet cable. don't know it there's anything that's been missed from your instruction. Please resolve this issue so I can move forward

cwnolt, 2023-04-23:
Does your cb1 have emmc?

Suggy50, 2023-04-23:
Yes I did all this. I resolved the issue by putting the CB1eMMc in aPI4B adaptor and configuring the the BootEnv and system files, put the CB1eMMc back into the M5P and finished setting it up. 
Now I have run into another issue. I set the CB1 eMMc as a second MCU but it wont open aline to the gpio chip

Suggy50, 2023-04-23:
as per here
https://github.com/bigtreetech/CB1/issues/112

cwnolt, 2023-04-23:
Well that I don't think I can help you with. Sorry

cwnolt, 2023-04-27:
They should be off to boot from emmc. You only put the usbotg and rpiboot on while flashing.

cwnolt, 2023-04-27:
Which board are you using?

jakeelee, 2023-04-27:
m8p, it wont show up as a flashdrive either even if i put the switches back to as when it was ok to flash

jakeelee, 2023-04-27:
think i found the problem. Did everything from the beginning and i forgot to change in the file that it should be the emmc version of cb1

Suggy50, 2023-04-27:
> think i found the problem. Did everything from the beginning and i forgot to change in the file that it should be the emmc version of cb1

I missed this 1st time around. I tried to use the gpio pins on the cb1 emmc version, configured the cb1 as host MCU b lineut unable open gpio line

jakeelee, 2023-04-28:
I have to check after work today but it looked like the file changed itself back to default when flashing memory.
hmmmmmmm

Sonnenwerk, 2023-06-08:
I started with installing CB1 emmc in MantaEZ3. Installed current version of image to SDCard, then changed fdtfile entry and also set the wifi in config. But it does not help. Any ideas ?
Can i see booting it up anywhere with flickering lights on the board ?

DP4US, 2023-06-25:
I am having the same issue. I can use the emmc as a USB device but not bootable. I changed the bootenv file to emmc from sd. However even if I have it set to eemc the sd will take over as advertised. I can access the emmc when pin otg and reboot pins are on with powershell command [btt manual]. I still cant get the emmc to boot on the sdcard. I do  not have the sd card in while trying to boot the emmc. its almost as if the setting is being ignored.

Hoot0118, 2024-05-16:
I followed every guide i can find and when trying to boot from the emmc it just stays stuck on btt logo and circle of death, I quaddroople checked dip switches, config files everything they said to uncomment is done, its set to emmc. pulling m y hair out here let me know what I need to provide I flashed the btt version to the emmc perfectly 10 times at this point just cannot get any further any help would be appreciated. 
[boardenv.txt](https://github.com/bigtreetech/CB1/files/15329558/boardenv.txt)

Hoot0118, 2024-05-17:
> Hi, how did you set the dip switches after flashing the emmc?

i reverted them to off as described in manual

cwnolt, 2024-06-23:
> I followed every guide I can find and when trying to boot from the emmc it just stays stuck on btt logo and circle of death, I checked dip switches, config files everything they said to uncomment is done, it is set to emmc. pulling my hair out here let me know what I need to provide I flashed the btt version to the emmc perfectly 10 times at this point just cannot get any further any help would be appreciated. edit: I did flash to sdcard as well and that portion works flawlessly I even got the btt eddy working but need the storage of the emmc for stability micro sds tend to corrupt when you need them most. [boardenv.txt](https://github.com/bigtreetech/CB1/files/15329558/boardenv.txt)

Do you have a monitor hooked up? I had that issue too but I could access it fine from a different computer via browser and I think via putty. Haven't had to have a monitor hooked up so haven't looked into it.

jakeelee, 2024-06-23:
> Do you have a monitor hooked up? I had that issue too but I could access it fine from a different computer via browser and I think via putty. Haven't had to have a monitor hooked up so haven't [...]

I guess you have removed the sdcard as this is highest in priority`?