bigtreetech/CB1 #111: CB1 - Only reachable via IPv6 not via IPv4

LsHallo, 2023-04-20:
I'm trying to connect to my CB1 via IPv4
I've got the IP address:

```
biqu@BTT-CB1:~$  ip a
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether aa:aa:aa:aa:aa brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.17/24 brd 192.168.178.255 scope global noprefixroute wlan0
       valid_lft forever preferred_lft forever
    inet6 2003::[snip]:b0ac/64 scope global dynamic noprefixroute 
       valid_lft 172794sec preferred_lft 86394sec
    inet6 fe80::[snip]:4ac5:6092/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```

When I then try: `ssh biqu@192.168.178.17` the connection times out.
`ssh biqu@[2003::[snip]:b0ac]` works fine.

I really can't explain this one...
When I connect ethernet I can reach the IPv4 of the wifi adapter. But i figure it then goes via LAN to reach the IPv4 on the CB1 and it's not magically fixed...

I have modified `/etc/NetworkManager/system-connections/MyWifi.nmconnection` to use a static IPv4 like so:
```
[...]
[ipv4]
dns-search=
method=manual
address1=192.168.178.17/24,192.168.178.1
dns=1.1.1.1;8.8.8.8
[...]
```

Thanks in advance!

LsHallo, 2023-04-20:
Solved...
I have removed the static IP from eth0 and now wlan0 is working as expected. Don't know why one influences the other...

closed 2023-04-20 by LsHallo

Avarios, 2023-05-24:
> 

How did you do that ?

LsHallo, 2023-05-24:
> How did you do that ?

I used network manager to edit my connections. You can use the terminal ui for that: `nmtui` and then navigate with arrow keys, tab and enter.