bigtreetech/CB1 #112: gpio pins NOT working with CB1 eMMc V1.1

Suggy50, 2023-04-22:
CB1 eMMc on Manta M5P
I have setup the CB1 eMMc as a second MCU configured gpio79 for PI-FAN which is the socket on the M5P. Settings have been verified by several people and all have said correct but I get this message MCU 'cb1' error during config: Unable to open out GPIO chip line

Please fix or report back if I have a hardware issue

Suggy50, 2023-05-01:
Still no resolution to this problem
Bigtreetech you need to fix this problem.
A little more information. When setting up MCU host I try to install "sudo apt-get install gpiod" but get error
"Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package gpiod"

Drew3164, 2023-06-03:
Is this fixed yet?

TristanScott44, 2023-06-06:
> Still no resolution to this problem Bigtreetech you need to fix this problem. A little more information. When setting up MCU host I try to install "sudo apt-get install gpiod" but get error "Reading [...]

I had this same issue and I ended up solving it.  I can't find the exact website I had referenced but I want to say all it took was updating the installed packages.  
https://raspberry-projects.com/pi/command-line/update-packages

Give it a try.

Grimones, 2023-07-21:
Hey guys
I'm in the same boat as you
Have a manta m8p v1.1 with CB1 eMMC and i'm not able to connect get the gpio26 to control the fan
I have installed klipper firmware for cb1, i've checked with `gpioinfo` that I have two `gpiochip` and nothing works.
If i set `CB1:gpio26`, `CB1:gpiochip0/gpio26` or `CB1:gpiochip1/gpio26` - nothing works.
Everything is up to date including system packages. Not sure into what direction I should look

Suggy50, 2023-07-22:
> Hey guys I'm in the same boat as you Have a manta m8p v1.1 with CB1 eMMC and i'm not able to connect to `gpio26` to control the fan. I have installed klipper firmware for cb1, i've checked with `gpioinfo` that I have two `gpiochip` and nothing works. If i set `CB1:gpio26`, `CB1:gpiochip0/gpio26` or `CB1:gpiochip1/gpio26` - nothing works. Same error `Unable to open out GPIO chip line` Everything is up to date including system packages. Not sure into what direction I should look

You need to change gpio26 to gpio269

Grimones, 2023-07-22:
> > Hey guys I'm in the same boat as you Have a manta m8p v1.1 with CB1 eMMC and i'm not able to connect to `gpio26` to control the fan. I have installed klipper firmware for cb1, i've checked with `gpioinfo` that I have two `gpiochip` and nothing works. If i set `CB1:gpio26`, `CB1:gpiochip0/gpio26` or `CB1:gpiochip1/gpio26` - nothing works. Same error `Unable to open out GPIO chip line` Everything is up to date including system packages. Not sure into what direction I should look
> 
> You need to change gpio26 to gpio269 These are my settings
> 
> [temperature_fan cb1_Fan] pin: cb1:gpio269 sensor_type: temperature_host control: watermark min_temp: 0 max_temp: 85 target_temp: 45

OMG
you are my saviour :) it worked like a charm.
And how should I calculate the gpio pin number? Let's say I want to use gpio23

Pneumanifest, 2023-07-22:
If you are using raspberry pi gpio numbers then that will not work, it's not the same SoC this is the key vor the gpio numbers, https://github.com/bigtreetech/CB1#40-pin-gpio

Grimones, 2023-07-24:
Thanks for that. I was thinking that in manta m8p docs in pinout image it's showing CB1 pinout rather than CM4. But now I understand it and can see in table if pinout says GPIO6 in CM4 column then it maps to GPIO265 in CB1 eMMC column.
I think the ticket can be closed now.

MrBlobfisch, 2025-03-22:
> You need to change gpio26 to gpio269 These are my settings
>
> [temperature_fan cb1_Fan] pin: cb1:gpio269 sensor_type: temperature_host control: watermark min_temp: 0 max_temp: 85 target_temp: 45

aight im opening up this old issue again lol
i have an btt m5p and cb1
gpio269 doesnt work for me, does anyone know why?

StackingLayers, 2025-04-21:
> aight im opening up this old issue again lol i have an btt m5p and cb1 gpio269 doesnt work for me, does anyone know why?

Because you are likely using a non eMMc (embedded storage) version. The regular SD card version uses gpio79 for that fan pin.