bigtreetech/CB1 #113: Wrong resolution on 7" screen

stenc55, 2023-04-23:
I have seen this question by numerous CB1 users and not a single answer from BTT.
I have Waveshare 7" touchscreen. In leaflet that came with the screen is written:
Append the following lines to config.txt:
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 1024 600 60 6 0 0 0 

Also, in CB1 install settings is this:
CB1 will automatically identify the HDMI resolution, but if your HDMI screen cannot report the resolution through the EDID normally, we can forcibly specify the resolution of CB1 output by uncomment extraargs=video and set the actual resolution.
For example:
BTT-HDMI7 resolution = 1024x600: extraargs=video=HDMI-A-1:1024x600-24@60

Nothing of that doesn't work. No matter what I do, I have the same result. Bottom of the screen is missing by approx 200 pixels.

Screen is not listed in xrandr output. Is there no driver installed at all?

eboston, 2023-06-05:
I am having the same issues.  CB1 does not seem to want to work with the Waveshare displays.  I did ask somewhere about this, but don't remember where, and got some responses back but can't find them now.  I ended up changing from the M5P board to and Octopus with a Raspberry Pi.  I have seen where Raspberry Pis are suppose to be getting back in stock in the next month or two so I will probably wait for the CM4 to become available again and switch the CB1 out.

TheDenS, 2023-06-09:
From manual, it work for me
### Configuration

Follow the manufacturer instructions on how to install your screen. In general if you see a white screen, then it's not properly installed, ensure that you at least see a console, Then [install](Installation.md) KlipperScreen, if you are having troubles refer to the [troubleshooting page](Troubleshooting.md) for further information.

## Screen rotation
Currently there is no support for rotation at runtime, do not use xrandr to rotate the screen.
Configure the server to start in the desired orientation, there are many ways to achieve this, here is one:

Create /usr/share/X11/xorg.conf.d/90-monitor.conf
```bash
sudo nano /usr/share/X11/xorg.conf.d/90-monitor.conf
```

Paste this section modifying the options to suit your needs:
```
Section "Monitor"
    Identifier "DPI-1"
    # This identifier would be the same as the name of the connector printed by xrandr.
    # it can be "HDMI-0" "DisplayPort-0", "DSI-0", "DVI-0", "DPI-0" etc

    Option "Rotate" "left"
    # Valid rotation options are normal,inverted,left,right

    Option "PreferredMode" "1920x1080"
    # May be necesary if you are not getting your prefered resolution.
EndSection
```
Save the file and restart KlipperScreen.

```bash
sudo service KlipperScreen restart
```