bigtreetech/CB1 #114: CB1 eMMC stuck in loading screen

tbs-trappy, 2023-04-29:
Hi,

I flashed the minimal debian using raspberry pi imager, then disabled the boot flag and rebooted the unit. Now the screen is stuck in a perpetual booting screen showing Bigtreetech logo and a loading animation. How can I proceed to boot into debian?

Pneumanifest, 2023-04-29:
Did you add needs_root_rights=yes to /etc/X11/Xwrapper.config (sudo nano /etc/X11/Xwrapper.config, add needs_root_rights=yes)

tbs-trappy, 2023-04-29:
How would I do that? The device doesn't open through SSH, and I can't enter the console. All I see is a loading screen

tbs-trappy, 2023-04-29:
[image]
[image]

Pneumanifest, 2023-04-29:
Oh I misread your post. I thought it was just klipper screen that wasn't booting. Why would you disable the boot flag? Those are normally set to direct to the boot parton to get things started. It's probably best to re-burn the image and don't disable the boot flag.

tbs-trappy, 2023-04-29:
You have to disable the bootloader jumper, otherwise you just get into the burn mode. I have re-burnt several times now, and I never get past this screen. I am not trying to use this for Klipper btw, I just want to run vanilla debian. I assume the minimal image is for that.

Pneumanifest, 2023-04-29:
Oh I see what you are saying now, the physical boot switch on that board to go from OTG to boot. I thought you were disabling a boot flag in the system. 
When you were done flashing, before taking it out of OTG mode, did you go into the BoardEnv.txt in the boot partition and edit the fdtfile line to read,  ```fdtfile=sun50i-h616-biqu-emmc``` ? You will also need to input your wifi info in the system.cfg file if you are planning to use that.

As for the screen, there's a chance that the CB1 doesn't support DIP interface. I definitely doesn't support DSI. I know that HDMI and SPI are supported though.

tbs-trappy, 2023-04-30:
thanks. I will try that right away, I did not change the fdtfile. and yes, the screen seems not to be compatible. I will need to wait for bigtreetech's CB1 screen to become available next week.

having high hopes that this can replace my CM4s. I just need a linux console and a USB driver. not using it for 3D printing.

Pneumanifest, 2023-04-30:
If you haven't seen it already, make sure to read about the gpio assignment in the user guide for the cb1. They are not the same numbers as the raspberry pi. There is a formula to use to get the correct gpio numbers.

tbs-trappy, 2023-05-05:
that fixed it, thanks!

closed 2023-05-05 by tbs-trappy