bigtreetech/CB1 #115: SPI screen and Klipperscreen stuck on login

Doormat1, 2023-05-02:
I have a CB1 and a manta e3 ez board and want to get Klipperscreen working on it. I have been to boardenv and uncommented the SPI screen part so my screen works and loads up the starting console but stops at the log in part. I check the logs and xlog states no screen found. What am I doing wrong?

hayesmp, 2023-05-22:
My board loads completely, but the screen remains white the entire time.

Pneumanifest, 2023-05-23:
@Doormat1 I would try to re-flash the image to the SD card. Redownload it to make sure nothing was corrupted the first time. With the latest CB1 full Klipper image all you should need to do is (@hayesmp do this) uncomment overlays=tft35_spi in the BoardEnv.txt.

hayesmp, 2023-05-23:
Does it make a difference that I have the emmc cb1 (that I’m unable to flash)?

Pneumanifest, 2023-05-23:
Doesn't matter if is on emmc or SD. They are especially the same thing. Not really the right topic here but in the same BoardEnv.txt file you need to set fdtfile to sun50i-h616-biqu-emmc to use the eMMc storage, and cannot have the SD in as that will take priority.

hayesmp, 2023-05-23:
Ok. I’ll try reflashing the sd card.

Tony-g80, 2023-06-20:
> My board loads completely, but the screen remains white the entire time.

im having the same issue with my TFT35 SPI, white screen no matter what, display enabled in boardenv.txt, spidev 1.2 enabled as suggested per docs, my btt pi is running latest V2.3.1 image and i can SSH into it and get to klipper landing page via browser but my screen is completely useless right now and i cannot ind any clues to fix it.

Doormat1, 2023-06-20:
Is your fpc correct? It needs to be the reverse kind, where the pads are on each end are on different sides of the cable. If it is the 
other type it will come up just white because the backlight is receiving power but nothing else I believe

Tony-g80, 2023-06-20:
> Is your fpc correct? It needs to be the reverse kind, where the pads are on each end are on different sides of the cable. If it is the
> other type it will come up just white because the backlight is receiving power but nothing else I believe

Thanks for the suggestion, I just double checked my doc cable and it is in the correct way, it’s the btt supplied one that came with the screen. 

After double checking each connectors orientation the cable is definitely in the right way, I spent most of yesterday trying to get this working but no luck.

Tony-g80, 2023-06-20:
> What cb1 version you you have installed? And what board is in on? All that needs enabling is the TFT35-spi overlay, although I have seen some say it didn't work until they used the TFT35-spi and mcp2515 overlay, that doesn't make sense but worth a try. Remember to sudo reboot so that the overlays are applied after editing (if done via ssh.)

Hi, I currently have the latest OS image V2.3.1,  I did enable the spi display setting in the boardenv.txt and also tried with and without the mcp2515 overlay and also with the spidev1.2 uncommented as recommended and also commented back out, the system is rebooted between changes but nothing seems to make any difference, the board is a btt pi v1.2 and it’s supposed to be straightforward but clearly somethings missing somewhere or it’s a dudd screen from the get go

Pneumanifest, 2023-06-20:
> Hi, I currently have the latest OS image V2.3.1,  I did enable the spi display setting in the boardenv.txt and also tried with and without the mcp2515 overlay and also with the spidev1.2 uncommented [...]

The BTT pi only needs the TFT35-spi overlay uncommented nothing more. The latest CB1 mainsail os works for me no issues. Make sure the blue tabs are facing the locking tabs for the ribbon cable.
Just to be sure run the ```sudo apt-get update && sudo apt-get upgrade -y``` command to make sure the os is up to date.

Tony-g80, 2023-06-20:
I’ve double checked my fpc cable orientation and it is in correctly both ends, I’m using the full size klipper loaded image provided by btt (1.02gb) also at some point I did try with just spi screen enabled but nothing.  

When I used a browser to get to landing page I was able to update klipper, mainsail ect as the tab was showing updates were available but still no change happened with the screen once everything was updated and rebooted.

Tony-g80, 2023-06-20:
Ok it seems we finally have success, put a new image on the sd card, the non klipper loaded image and enabled spi display and it works. 

The issue that needs to be cleared up on the documentation is where it tells you to set it up like the cb1 and that info itself states that aside from spi display being uncommented that spidev1.2 is needed as well, the setup docs maybe need to include a note stating those parts are not needed for btt pi board, either way it’s working now so thank you to all for suggestions and assistance. 

Interestingly I loaded the basic image after eventually finding another post online where the op had to do the same after the fully loaded klipper image caused the same issue, is their something in their that btt may need to look at?

Pneumanifest, 2023-06-20:
> The issue that needs to be cleared up on the documentation is where it tells you to set it up like the cb1 and that info itself states that aside from spi display being uncommented that spidev1.2 is [...]

Not sure what you are referring to, the user guide shows to use only the TFT35-spi overlay and nothing about spi1.2 that is for other things like an adxl, the full Klipper loaded image worked perfectly with my BTT pi and TFT35-spi nothing more needed than using the single overlay as stated. https://github.com/bigtreetech/TFT35-SPI/blob/master/v2/BIGTREETECH%20TFT35%20SPI%20V2.1%20User%20Manual.pdf