bigtreetech/CB1 #117: Multicam

darxx1, 2023-05-06:
I am trying to run two cameras on mainsail. 
Unfortunately, the first camera reports "no signal", although you can see the changing fps.
Everything is well configured but one camera is not working. If I disconnect the working one, the other starts working. I don't know what else I can do.

[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true

[cam 1]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/v4l/by-id/usb-GENERAL_2K_HD_Camera-video-index0                     # See Log for available ...
resolution: 800x600                     # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

[cam 2]
mode: mjpg                              # mjpg/rtsp
port: 8081                              # Port
device: /dev/v4l/by-id/usb-Sunplus_IT_Co_Trust_USB_Camera_20200707002-video-index0                    # See Log for available ...
resolution: 800x600                     # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.

oMTEo, 2023-05-20:
I have exact the same Pproblem.

kwalters3, 2023-05-21:
I had a similar problem with using a Logitech C615 and C920 camera with mine while trying to use max camera settings.  I think the problem is there isn't enough bandwidth on the USB bus.  Got both of my cameras working by testing other available resolution combinations from the cameras

oMTEo, 2023-05-22:
Ive tried that now with 2 identical Cams (USB-Endoscopes), played with resolutions 320x240, 640x480 ... still same bug, only 1 cam shows up, on the other shows black background and "no signal", with fluctuating fps...

kwalters3, 2023-06-02:
@oMTEo What does your crowsnest.conf and crowsnest.log files look like?

The log file will tell you some information about why the camera is failing.  If disconnecting the working camera and the other starts displaying video it is a issue with the usb not having enough bandwidth so support both cameras feeds

oMTEo, 2023-06-02:
[crowsnest.conf.txt](https://github.com/bigtreetech/CB1/files/11634896/crowsnest.conf.txt)
[crowsnest.log](https://github.com/bigtreetech/CB1/files/11634897/crowsnest.log)