bigtreetech/CB1 #118: CB1 eMMC won't flash (E3EZ Manta board)

hayesmp, 2023-05-10:
I can get through to step 7. The volume mounts but when I try to write the image, I got a USB error. I've tried both Raspberry Pi Imager and Balena Etcher.

I have the jumpers 3 RPIBOOT and 4 USBOTG and the switch set to USB.

hayesmp, 2023-05-11:
Replace the driver with the WinUSB driver? using Zadig. Yes.

Tina-Builds, 2023-05-11:
Sorry, just to make sure. I had no problems writing the OS to the CB-1 16G eMMC on the E3EZ. 
Sometimes I had problems with other devices (e-cig battery mods) to write the image. Could you use a USB 2.0 port (on the PC) or a different USB A to µUSB cable? If you have no USB 2.0 port, coil you try a USB 2.0 hub? This did solve a upload problem for me flashing a custom FW to a Lost Vape Grus mod.

Wuschfee, 2023-05-11:
CBI 1 with 32 GB eMMC v1.1 what I have received  2-3 weeks ago.. Tried on my PC ans freshly Installed Laptop(win 10, USB2 and USB 3) to disclose any driver issue.
I  can't even see the EMMC..... If the MS OS recognize it, the drive can not be accessed, (the one or 2  time when it appeared) by formatting is jut dismounting it self.  If OS recognized the drive, could not assign any letter to the drive. Tried with SD card to boot the CBI 1. Booting from SD card it has hanged at the BTT logo (i don't think loading the firs boot should not take 45-60 min.... then I have gave up)... I bought an extra Waveshare IO board to verify the card. And its dead in the water.. OS recognize the base CB1 with the Manta Mp8 and with the Waveshare IO. ether way, eMCC can not be accessed, and it wont boot from SD card ether.....
Opening Issue ticket did don't work ether.. No verifying email received.. 
Facebook Official community did not replayed me ether.
And yes I was forced to by a raspberry CM4  4GB 8eMMC, and Manta m8P is now in a state of configuration...
Regarding the CM1 32Gb eMMC v 1.1.  without any replay from  BiQu  or Contact to them, no quit shore how preside with the card...

NaMoUrGCh, 2023-09-29:
I have practically the same problem, after installing uboot it is impossible to mount the eMMC like a usb disk

NaMoUrGCh, 2023-09-29:
I managed to install Linux on the CB1 using a Rasperry PI.

First, subxi-fel provided by BTT did not work for me, I had to make a clone of https://github.com/linux-sunxi/sunxi-tools and follow https://github.com/linux-sunxi/sunxi-tools#building
And then `sudo ./sunxi-fel -v uboot ./u-boot-sunxi-cb1-emmc.bin` works (-v for verbose)
This command have to be done every time you want to access to the eMMC like USB disk

Second, as I use my PI without desktop, I use command `dd` to burn the image on the eMMC thanks to generic tutorial: https://linuxconfig.org/burning-cds-over-the-internet-with-or-without-iso-image
In summary :
`sudo fdisk -l` to define where eMMC is plug
`sudo dd bs=4M if=/path/to/file.img of=/dev/sdX status=progress oflag=sync` to burn the image to the eMMC
`sudo mount /dev/sdX1 /mnt ` to access the files system.cfg and BoardEnv.txt like `sudo nano /mnt/system.cfg`

Hope it's help you.

Hoot0118, 2024-05-15:
> I managed to install Linux on the CB1 using a Rasperry PI.
>
> First, subxi-fel provided by BTT did not work for me, I had to make a clone of https://github.com/linux-sunxi/sunxi-tools and follow [...]

Your first step swapping out the sunxi-tools saved me from loosing my mind once i did that everything finally showed up your a hero!