bigtreetech/CB1 #121: CB1+Pi4B+Octoprint+TFT7"+Octo Screen.???????

Aleks-MO, 2023-05-18:
**CB1+Pi4B+Octoprint+TFT7"+Octo Screen.**
CB1+Pi4B+Octoprint+TFT7"+Octo Screen.
Who can help? Two questions, what is missing in the description of 'linux-u-boot-h616-current' when installing oktoscreen?
There are no button icons in oktoscreen, the library "librsvg" is not installed. 
[image]

Pneumanifest, 2023-05-18:
The no discription warning is just letting you know that there is no discription for that package. You can add, 
    Description: and some sort of description
And it will stop that warning

Aleks-MO, 2023-05-18:
@Pneumanifest  Thanks, I'll try. 
Tell me what to do with library "librsvg", there are no icons in OctoScren.((((

Pneumanifest, 2023-05-18:
It's a library for turning SVG files into raster images. If you don't have it you should probably install it. That's probably why the icons are not showing.

Pneumanifest, 2023-05-18:
Try,

    ```sudo apt update```
    ```sudo apt install librsvg2-2```

Aleks-MO, 2023-05-18:
The installation has passed, after the reboot, the button icons are empty... ((((
biqu@BTT-CB1:~$ sudo apt install librsvg2-2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librsvg2-2 is already the newest version (2.50.3+dfsg-1).
The following packages were automatically installed and are no longer required:
  libdrm-amdgpu1 libdrm-nouveau2 libdrm-radeon1 libdrm2 libegl-mesa0 libegl1 libfontenc1
  libgbm1 libgl1 libgl1-mesa-dri libglapi-mesa libglvnd0 libglx-mesa0 libglx0 libinput10
  libllvm11 libmtdev1 libpciaccess0 libsensors5 libunwind8 libvulkan1 libwayland-server0
  libx11-xcb1 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-sync1
  libxcb-xfixes0 libxfont2 libxshmfence1 libz3-4
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 34 not upgraded.

Aleks-MO, 2023-05-18:
@Pneumanifest 
[image]

Pneumanifest, 2023-05-19:
Are you sure the icons needed are installed?

Aleks-MO, 2023-05-19:
Icons are there, how to check linked correctly?
[image]

Pneumanifest, 2023-05-19:
Can you show the output of this,
    ls -la /opt/octoscreen/styles/z-bolt/images

Aleks-MO, 2023-05-19:
biqu@BTT-CB1:~$ ls -la /opt/octoscreen/styles/z-bolt/images
total 552
drwxr-xr-x 4 root root  4096 May 19 08:58 .
drwxr-xr-x 3 root root  4096 May 19 08:58 ..
-rw-r--r-- 1 root root  5429 Aug  8  2021 actions.svg
-rw-r--r-- 1 root root  1276 Aug  8  2021 backspace.svg
-rw-r--r-- 1 root root  3171 Aug  8  2021 back.svg
drwxr-xr-x 2 root root  4096 May 19 08:58 bed-level-parts
-rwxr-xr-x 1 root root  3072 Aug  8  2021 bed-level.svg
-rw-r--r-- 1 root root  3710 Aug  8  2021 bed.svg
-rwxr-xr-x 1 root root   489 Aug  8  2021 blank.svg
-rw-r--r-- 1 root root  1424 Aug  8  2021 complete.svg
-rw-r--r-- 1 root root  4441 Aug  8  2021 control.svg
-rw-r--r-- 1 root root  2966 Aug  8  2021 cool-down.svg
-rw-r--r-- 1 root root  4231 Aug  8  2021 custom-script.svg
-rw-r--r-- 1 root root  3057 Aug  8  2021 decrease.svg
-rwxr-xr-x 1 root root  6147 Aug  8  2021 extruder-extrude.svg
-rwxr-xr-x 1 root root  4190 Aug  8  2021 extruder-multi.svg
-rwxr-xr-x 1 root root  6150 Aug  8  2021 extruder-retract.svg
-rwxr-xr-x 1 root root  3584 Aug  8  2021 extruder.svg
-rw-r--r-- 1 root root  3115 Aug  8  2021 extruder-typeB-1.svg
-rw-r--r-- 1 root root  3123 Aug  8  2021 extruder-typeB-2.svg
-rw-r--r-- 1 root root  3123 Aug  8  2021 extruder-typeB-3.svg
-rw-r--r-- 1 root root  3123 Aug  8  2021 extruder-typeB-4.svg
-rw-r--r-- 1 root root  2821 Aug  8  2021 extruder-typeB.svg
-rw-r--r-- 1 root root  8095 Aug  8  2021 fan-off.svg
-rw-r--r-- 1 root root  8576 Aug  8  2021 fan-on.svg
-rw-r--r-- 1 root root  8576 Aug  8  2021 fan.svg
-rwxr-xr-x 1 root root  7960 Aug  8  2021 filament-spool-load.svg
-rwxr-xr-x 1 root root  7537 Aug  8  2021 filament-spool.svg
-rwxr-xr-x 1 root root  7962 Aug  8  2021 filament-spool-unload.svg
-rwxr-xr-x 1 root root 10397 Aug  8  2021 file-gcode.svg
-rw-r--r-- 1 root root  4503 Aug  8  2021 files.svg
-rwxr-xr-x 1 root root  4165 Aug  8  2021 file-stl.svg
-rw-r--r-- 1 root root  2145 Aug  8  2021 folder.svg
-rw-r--r-- 1 root root  3289 Aug  8  2021 heat-up.svg
-rw-r--r-- 1 root root  3204 Aug  8  2021 home.svg
-rw-r--r-- 1 root root  3578 Aug  8  2021 home-x.svg
-rw-r--r-- 1 root root  3640 Aug  8  2021 home-y.svg
-rw-r--r-- 1 root root  3562 Aug  8  2021 home-z.svg
-rwxr-xr-x 1 root root  4617 Aug  8  2021 hotend-1.svg
-rwxr-xr-x 1 root root  4617 Aug  8  2021 hotend-2.svg
-rwxr-xr-x 1 root root  4617 Aug  8  2021 hotend-3.svg
-rwxr-xr-x 1 root root  4617 Aug  8  2021 hotend-4.svg
-rwxr-xr-x 1 root root 10409 Aug  8  2021 hotend-multi.svg
-rwxr-xr-x 1 root root  4342 Aug  8  2021 hotend.svg
-rw-r--r-- 1 root root  3105 Aug  8  2021 increase.svg
-rw-r--r-- 1 root root  2063 Aug  8  2021 info.svg
drwxr-xr-x 2 root root  4096 May 19 08:58 logos
-rw-r--r-- 1 root root  1619 Aug  8  2021 magnet-off.svg
-rw-r--r-- 1 root root  7757 Aug  8  2021 magnet-on.svg
-rw-r--r-- 1 root root  1834 Aug  8  2021 motor-off.svg
-rw-r--r-- 1 root root  1980 Aug  8  2021 motor-on.svg
-rw-r--r-- 1 root root  7316 Aug  8  2021 move-step.svg
-rw-r--r-- 1 root root  4741 Aug  8  2021 move.svg
-rw-r--r-- 1 root root  3247 Aug  8  2021 move-x+.svg
-rw-r--r-- 1 root root  3284 Aug  8  2021 move-x-.svg
-rw-r--r-- 1 root root  3197 Aug  8  2021 move-y+.svg
-rw-r--r-- 1 root root  3237 Aug  8  2021 move-y-.svg
-rw-r--r-- 1 root root  3201 Aug  8  2021 move-z+.svg
-rw-r--r-- 1 root root  3204 Aug  8  2021 move-z-.svg
-rw-r--r-- 1 root root  2803 Aug  8  2021 network.svg
-rwxr-xr-x 1 root root  2098 Aug  8  2021 nozzle-1.svg
-rwxr-xr-x 1 root root  2098 Aug  8  2021 nozzle-2.svg
-rwxr-xr-x 1 root root  2098 Aug  8  2021 nozzle-3.svg
-rwxr-xr-x 1 root root  2098 Aug  8  2021 nozzle-4.svg
-rwxr-xr-x 1 root root  1871 Aug  8  2021 nozzle.svg
-rwxr-xr-x 1 root root  1868 Aug  8  2021 octoprint-tentacle.svg
-rwxr-xr-x 1 root root   897 Aug  8  2021 open.svg
-rw-r--r-- 1 root root  3465 Aug  8  2021 pause.svg
[... 14 lines cut ...]
-rw-r--r-- 1 root root  3105 Aug  8  2021 speed-34%.svg
-rw-r--r-- 1 root root  3097 Aug  8  2021 speed-50%.svg
-rw-r--r-- 1 root root  3108 Aug  8  2021 speed-67%.svg
-rw-r--r-- 1 root root  3109 Aug  8  2021 speed-83%.svg
-rw-r--r-- 1 root root  2967 Aug  8  2021 speed-fast.svg
-rw-r--r-- 1 root root  2969 Aug  8  2021 speed-normal.svg
-rw-r--r-- 1 root root  2973 Aug  8  2021 speed-slow.svg
-rw-r--r-- 1 root root  5867 Aug  8  2021 status.svg
-rw-r--r-- 1 root root  3104 Aug  8  2021 stop.svg
-rwxr-xr-x 1 root root   925 Aug  8  2021 time.svg
-rw-r--r-- 1 root root  7936 Aug  8  2021 tool-changer.svg
-rwxr-xr-x 1 root root  3517 Aug  8  2021 toolhead-typeA-1.svg
-rwxr-xr-x 1 root root  3518 Aug  8  2021 toolhead-typeA-2.svg
-rwxr-xr-x 1 root root  3518 Aug  8  2021 toolhead-typeA-3.svg
-rwxr-xr-x 1 root root  3520 Aug  8  2021 toolhead-typeA-4.svg
-rwxr-xr-x 1 root root  3808 Aug  8  2021 toolhead-typeA.svg
-rwxr-xr-x 1 root root  2321 Aug  8  2021 toolhead-typeB-1.svg
-rwxr-xr-x 1 root root  2320 Aug  8  2021 toolhead-typeB-2.svg
-rwxr-xr-x 1 root root  2320 Aug  8  2021 toolhead-typeB-3.svg
-rwxr-xr-x 1 root root  2320 Aug  8  2021 toolhead-typeB-4.svg
-rwxr-xr-x 1 root root  2351 Aug  8  2021 toolhead-typeB.svg
-rw-r--r-- 1 root root  2325 Aug  8  2021 z-calibration.svg
-rw-r--r-- 1 root root  3982 Aug  8  2021 z-offset-decrease.svg
-rw-r--r-- 1 root root  3986 Aug  8  2021 z-offset-increase.svg
biqu@BTT-CB1:~$

Pneumanifest, 2023-05-19:
Maybe try stoping and restarting the service 
    
    sudo systemctl stop octoscreen.service
    sudo systemctl start octoscreen.service

Or you may need to do an uninstall reinstall. It's been a while since I've used octoscreen, but it looks like it's hasn't been updated since I used it last.

Pneumanifest, 2023-05-19:
Maybe you can give octodash a go instead https://unchartedbull.github.io/OctoDash/index.html

Aleks-MO, 2023-05-19:
Stopping and restarting oktoscreen, did not help, reinstalled and put it in "clean", everything is the same ...
Yes, it looks like the project died, judging by the number of questions ...((

Pneumanifest, 2023-05-19:
Yeah it hasn't been updated for 2 years

Aleks-MO, 2023-05-19:
Is it possible to replace svg images with png, and register ....?

Aleks-MO, 2023-05-19:
@Pneumanifest       Ha, ha, dripped deeper.... how true is this? 32 and 64 bits...          https://github.com/Z-Bolt/OctoScreen/issues/288