bigtreetech/CB1 #123: imposible rotate screen

jarvis710, 2023-05-27:
Everything I have found on the internet, refers to the Rasberry Pi3 or 4 and not to the CB1, I do not see how to rotate the screen. Supposedly there is an option in system.cfg to do it but nothing. I listen to suggestions; I've been looking all over the internet for several days and nothing.

jarvis710, 2023-05-27:
correction, the system.cfg does work; but not that, it rotates the touch so, not the image

bthubs, 2023-05-30:
I'm having this same issue with the TFT35 SPI and CB1. I've tried a lot of things already including referring to KlipperScreen documentation, changing params that work on RPi, etc. Only thing I managed to do was rotate the transformation matrix ie. touchscreen registers as if it is inverted (correct orientation) but the display is upside down. I thought maybe the "identifier" was incorrect but I've also ran thru many different common identifiers by trial and error and still no display rotation. What I've concluded thus far is that the display is neither handled by X11 or any of the params that would work for an RPi, but the touchscreen is.

bthubs, 2023-06-06:
> THANKS

I figured out how to rotate my btt tft35 SPI. If you’re using an SPI screen, this should work for you. I detailed it on the btt tft35 SPI GitHub. 

`https://github.com/bigtreetech/TFT35-SPI/issues/7 `

Good luck!

mainmachine, 2025-05-29:
I realize this is an old issue, however there is a much easier solution than modifying device tree files. I tested this on a 7" HDMI touchscreen, so YMMV with SPI or other screens, though if your screen shows up when you run `xrandr`, then it will probably work:

```
biqu@bigtreetech-cb1:~$ DISPLAY=:0 xrandr
Screen 0: minimum 320 x 200, current 1024 x 600, maximum 8192 x 8192
HDMI-1 connected primary 1024x600+0+0 (normal left inverted right x axis y axis) 150mm x 90mm
   1024x600      59.85*+
   800x450       60.05  
```

You can rotate the display output like so:

```
DISPLAY=:0 xrandr --output HDMI-1 --rotate left
```
In my case, I had to rotate `left`. Changes are immediate so you can try each one to see what works for you. Options are as seen above: `normal left inverted right`

Changing settings with `xrandr` are not saved when the X server restarts. To make those changes persistent, follow the KlipperScreen "Universal xorg configuration" here: https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/Rotation/