bigtreetech/CB1 #125: OS Crashing and Corrupting SD card

420noscope-exe, 2023-06-13:
I have a manta M8P mainboard with a CB1 installed onto it, running the provided klipper image. I always leave my machine idling, powered on at all times. Randomly, after about a week of idling, the web interface crashes. I am unable to access it at all.
I cannot ssh into the device either. The only solution is to unplug the printer and plug it back it. This has a high probability of corrupting my SD card, and I've had to re-image it about 4 times now. Where do I start with this?

420noscope-exe, 2023-06-13:
#105 https://github.com/bigtreetech/CB1/issues/105#issuecomment-1538412749

this user experienced a very similar occurrence.

klipperbot, 2023-07-25:
Same thing is happening to me as well. The printer runs fine when I reflash the SD card. But as soon as I turn it off it just stays on the boot screen and the CB1 light never turn back on. Like it's not getting power sent to it.

Fliptronics, 2023-07-25:
I have the same problems with M5P + CB1 board. After a couple of restarts the SD card looks like getting corrupt. Re-flashing solve the problem for a while.  What is the best size and brand for the SD-card?

klipperbot, 2023-07-25:
@Fliptronics I have used both 8gb and 16gb it does the same thing with both. I have read that the max is anything under 64gbs.

frutabruta, 2023-07-25:
A1 or A2 certified microSD should do the trick. These are made to handle frequent reads/writes in Android phones.

Pneumanifest, 2023-07-25:
> 

Make sure that everything is up to date with the OS 
    sudo apt-get update && sudo apt-get upgrade -y
Also use a minimum of 16GB high-quality high speed SD (class 10, U1 minimum)

klipperbot, 2023-07-26:
[image]
This is what pops up on mine

Fliptronics, 2023-07-26:
I get the same and after a few more restart he hangs on the start-up logo.
I order a Kingston SDCIT2/16GBSP microSD card. Maybe that helps.

Pneumanifest, 2023-07-26:
Just want to point out that not shutting down the machine before powering off will eventually cause corruption of the file system. Just from reading the comments it seems that is what is going on.

Fliptronics, 2023-07-26:
I always do a shutdown in the klipper menu. But when the system hangs at start-up you can only cycle the power because no connection is possible, klipper hasn't started yet etc.

klipperbot, 2023-07-26:
Same no matter which way I turn it off it does the exact same thing. I just got it working again last night as of right now I'm just not going to turn it off until I fix the internet issue it seems like it's having.

MarkH9, 2023-07-29:
I have the same issue. Used Balena Etcher this morning to redo image and it worked until I turned printer off. Now again can't find it on the WiFi network. Will get anew sd card tomorrow and try again.

jcrienen, 2023-08-14:
I have the exact same issue. Sent BiQu an email today. I am using a SanDisk Extreme 32 GB A1 U3 MicroSD card which is brand new. The screen freezes and the OS crashes randomly. Right now it is stuck in the loading screen. Has anyone found a solution yet?

420noscope-exe, 2023-08-18:
> Just want to point out that not shutting down the machine before powering off will eventually cause corruption of the file system. Just from reading the comments it seems that is what is going on.

Well yes, but there is a critical process that is dying or failing to start correctly, forcing our hand to pull the power and not shut down the OS properly.  The web interface will not connect, and SSH fails to connect.  The only other option to reboot is a hard shutdown, there are no power buttons on the manta boards.  Updating the OS had no improvement on stability (like come on, that's the first thing I tried lol).

> A1 or A2 certified microSD should do the trick. These are made to handle frequent reads/writes in Android phones.

I am currently now testing a Sandisk ImageMate 32GB microSD card, rated UHS-I and A1.  I have been running it since Aug 2 with no issues so far (today is Aug 18).  More testing is needed to see if this fixed the problem.  My old SD card would crash roughly every 30 days.

420noscope-exe, 2023-09-13:
Ok, so after about a month of testing with the new SD card, I have yet again, lost communication with my printer via the network.  Going to 192.168.56 fails to load the page.  The printer was powered  and on between Aug 2 and Sep 11(when I noticed a problem), about a month like I have noticed previously.  I cannot ping this address locally, or SSH into it either, I do not get a response back.  I reserved this address for my printer in my router's settings, so it should not change.  Just to be sure, I logged into my router and I do not see the CB1 listed in attached devices.  I will attempt to attach a screen to the manta m8p board to see what is going on, and investigate further.

@jcrienen Have you had any luck with customer support or the new SD cards(Sandisk extreme vs. Industrial)?

jcrienen, 2023-09-13:
@420noscope-exe the 16 GB industrial SD card worked fine for about a week but then suddenly didn't boot anymore. The green light on my Manta M5P did not light up and the activity light did not blink. I tried several different older versions of the OS, but no luck. Ended up disconnecting all cables, powering via USB and still no luck unfortunately. After this, I removed the heatsink from my CB1 and reconnected it to the board which for some reason worked. I am pretty certain the board was connected securely before because it was working so I think the heatsink might have something to do with it (causing a short somewhere?). Today it ran fine except for one boot which did not load the OS succesfully. Tried a second time and the OS booted fine. Will update again if the board fails again.

420noscope-exe, 2023-10-12:
I have not touche my printer since it crashed and became unresponsive.  Today I finally attached a monitor to my manta m8p.  I tried hdmi 0 and hdmi 1.  I did not get a signal or picture on the monitor.  

@jcrienen Has reseating the board and heatsink permanently fixed the issue?

@bigtreetech Are you able to look into this at all?

jcrienen, 2023-10-12:
> I have not touche my printer since it crashed and became unresponsive. Today I finally attached a monitor to my manta m8p. I tried hdmi 0 and hdmi 1. I did not get a signal or picture on the [...]

I ended up replacing the CB1 with a CB1 eMMC which did work without any problems. The CB1 still works with the PI4B adapter without any issues, so I guess it has something to do with the SD card slot on my Manta M5P.

Abyss8, 2023-10-19:
I have problem to connect browser and putty, and when i go to wifi settings on Pad 7 it freezes.

DzmitryShamal, 2023-12-22:
I have the same problem. 
Expected that you could restore your files throught software that working with disk portations.
[image]

Fliptronics, 2023-12-22:
I am using a Kingston Industrial SDCIT2/16GBSP microSDHC card for a couple of months now without any problems so far. So good quality is solving a lot of problems.

lamtranBKHN, 2024-03-08:
Hi guys, I had the same issue with you, my CB1 on M8P doesnt want to boot up after few restart. I had read this thread plenty times and try to install all offical images on several SD cards. But the issue is still there. After spending my whole day to figure it out. It turns out, the issue mainly because of the image that came from BTT. I switch to Armbian image, and it works like a charm, reliable and fast. I currently use Armbian for a week without any issue ( everything work include the status led on M8P). I try to hot-unplug the power from the board all the time I turn my printer off but it's still working :D. If anyone has this issue can try [Armbian for CB1](https://www.armbian.com/bigtreetech-cb1/). Hope this help!

Pneumanifest, 2024-03-08:
@lamtranBKHN it's good to see that armbian is working well, bit one thing, you should never pull the power/turn off the printer before properly shutting down the system. The CB2 is a Linux computer and is always doing something in the background. Removing power without shutting down fist is a great was to corrupt the image. Even if it has been fine before doesn't mean it's a good thing to keep doing.

lamtranBKHN, 2024-03-13:
@Pneumanifest I edited my comment. I did this just for testing the Armbian image and everyone should not follow and apply this to a normal SCB aswell as Klipper powered printer :D

JuanSebastianOtalora, 2024-07-22:
I do have the same issue. Been going back and forth with Biqu support (as expected not very useful). 
I'm providing more details and behavior of the board.

**Problem: Manta v2.0 + CB1 v2.2 failing to boot after normal shutdown/reboot procedure, also reporting wrong frequency and bootloader issue for klipper.**

I'm going to recapitulate the whole issue and process including as much detail as possible.

I have two Ender 3 Pro, and 1 Ender 5 Plus.

Did the standard procedure in the manual that I got from BTT/github for manta v2.0, setting up the jumpers, flashing sd card, doing the wifi stuff... installed the board in the Ender 5 Plus (so far it was a stock printer). That same week after the tuning process for the speed, retraction and other values for printing I noticed the that when using the shutdown button in mainsail, the board would randomly leave the ACT LED on, no matter how much time I waited it would remain in that state, which I found odd because whenever that happened after power cycling it would take longer and longer to start, random file corruption would happen (in my case moonraker corrupted files), first though was the SD was bad, so I went ahead and replaced the SD card. The issue got even worse because now there were no corrupt files but the system would never start, that would happen almost always when the ACT LED remained on after shutting down and then power cycling ACT LED would never turn on again (LED was not damaged, just showing that system was not going up?), Ethernet LEDs either. So naturally I replaced the SD card brand. Same issue. Then I replaced the board with the other v2.0 using the sd cards that didn't work. and the same thing, no boot, no ACT LED on.

Purchased multiple packs of 8,16,32 and 64gb sd cards from, Adata, Kingston, Samsung, Patriot. No luck, every single one of those stopped working after sometime with the same issue. In both boards
Next thing I did was swap the CB1s on those v2.0 hoping that I could be that it was not connected properly or a CB1 was defective, no luck.
Then used those SD cards (that didnt show to start) in the manta 1.1 but the system never loaded again (I checked the CB1 versions and they are v2.2 including the CB1 in the 1.1 manta). The next logical step was to use the SD card from the manta 1.1 on the v2.0 boards, first boot no issue, I updated the system through mainsail, it did the standard automated reboot (ACT LED stuck on, then turned off and never turned on again, Ethernet didn't flash either). Power cycled and the system did not start as usual with that issue.

My thought was the problem was the BTT image, tried the minimal version, after updating and installing klipper throught KIAUH, the issue struck again, as @lamtranBKHN said armbian does work flawlessly (altho for my case the is no video output at all) and now boot/startup issues happen. I was told to try the RatOS system image. Joined RatOS discord server to get some guide on how to because the board was on the Development state for the image. 
During the process we found out that the board was not flashing klipper in the MCU, due to a problem with the bootloader, so we followed the steps in the Bootloader section for the manta 2.0 in BTT/github. A lot of problems popped up in the process.
If you have access to RatOS discord here is the link for that thread: https://discord.com/channels/582187371529764864/1206790356062969856/1206790356062969856
Amongst the issues we had but not limited to, board not getting the new bootloader, permission issues, and wrong freq information for the MCU (still a problem to this day, user that helped in that thread also had the same issues with the bootloader however his board does report the correct freq 500Mhz, for me still reports 400Mhz after the new bootloader)
[image]

Weirdly enough, I got myself an app to send SSH commands through the phone, and using "sudo shutdown now -P", the issue stopped happening when using the mainsail shutdown/reboot button, until I updated the system again in mainsail.

Once again i'm attaching the videos showing normal operation after shutdown and then stopping to start after second shutdown.
In my observation
**Normal startup procedure / behavior:**
PWR LED ON -> Ethernet flash sequence (whether is connected via wifi or ethernet) -> another Ethernet flash sequence  ->ACT LED blink.
**Normal power down procedure / behavior :**
Mainsail/SSH command for shutdown/reboot -> Mainsail reports disconnect -> SSH session "Remote side unexpectedly closed network connection / Session stopped" ->  ACT LED blink -> ACT LED OFF -> PWR LED remains on until psu switch is used.

**Abnormal startup procedure / behavior:**
PWR LED ON -> Ethernet flash sequence not present (whether is connected via wifi or ethernet) -> another Ethernet flash sequence not present ->ACT LED remains OFF. System does not boot, no mainsail no SSH
Just in case, the issue is also present when using ethernet.
**Abnormal power down procedure / behavior:**
Mainsail/SSH command for shutdown/reboot -> Mainsail reports disconnect -> SSH session "Remote side unexpectedly closed network connection / Session stopped" ->  ACT LED ON (solid no blinking) -> PWR LED remains on until psu switch is used. 

You can see the abnormal power down and startup in the following videos:

Video 2 shows the normal startup behavior
[ VIDEO 2.mp4](https://drive.google.com/file/d/1pG4SihyHn8Bn6YPEhXVriPXsRmF6FAhn/view?usp=drive_web)
After shutting down though mainsail, the board shows abnormal power down behavior
[ VIDEO 3.mp4](https://drive.google.com/file/d/1WDIo-qVwMTyphmGc0ujIHxVId4NmhgTX/view?usp=drive_web)
Both videos were recorded one after the other.

As you can see board did not boot at all. Using HDMI to see any video output doesn't work in any of the two mantas v2.0.

Does anybody here have the frequency issue too?