bigtreetech/CB1 #130: Plugging in USB flash drive causes dmesg log to flood with messages about data transfer.

cdsmith69, 2023-07-19:
With the new CB1 images dated 20230712, version V2.3.3, plugging in a USB flash drive floods the dmesg log with kernel messages about data transfer.  There is a large block of messages that appear when the drive is plugged in, then every two seconds from there on another block of messages is added, until the flash drive in removed.   The drive seems to work fine, but this reduces the usefulness of the dmesg log when everything gets pushed out by the flood of messages in only a few seconds.

I see the issue with both CB1_Debian11_Klipper_kernel5.16_202300712.img.xz and CB1_Debian11_minimal_kernel5.16_20230712.img.xz, on both a BTT Pi and a Manta M5P with a CB1.  I have tried several flash drives from different manufacturers and checked the drives have no file system errors.

Attached is a file where I redirected "dmseg -w" to the file and it saved over 6300 lines in only a few seconds with the flash drive plugged in.
[dmesglog.txt](https://github.com/bigtreetech/CB1/files/12096561/dmesglog.txt)

chimpnz, 2023-07-28:
Sounds like USB Storage Debug is turned on in the kernel. Same issue is described here:
https://www.linuxquestions.org/questions/gentoo-87/var-log-messages-has-endless-usb-storage-entries-788995/

bigtreetech, 2024-03-07:
Disabled in https://github.com/bigtreetech/CB1-Kernel/commit/53e0bce246366bc14a195ff6f660085d250c77c3
Thanks

closed 2024-03-07 by bigtreetech