bigtreetech/CB1 #132: Screen resolution

mcbeis, 2023-07-24:
Hi, 
I use the manta m8p with the CB1 and the waveshare 5" hdmi display with 800x480 pixel.

The BoardEnv.txt file contains:
extraargs=video=HDMI-A-1:800x480-24@60

and at boot the resolution is accordingly, but when klipperScreen starts, the resolution switchs to 1280x760 and the menu does not fit the screen.

Where to setup the proper resolution of 800x480?
Thanks!

Bernie-McGee, 2025-05-21:
I may be having a similar issue. Is your resolution stable once it switches to 1280x760? My resolution jumps around between being too high and correct.

Also, I believe BoardEnv.txt only controls resolution during boot. Once X11 starts for KlipperScreen, it should automatically detect the correct resolution. If its not, try setting the resolution in X11 config.

mcbeis, 2025-05-23:
Hi Bernie,
In my case, the resolution did not jump around, but in the meantime, I have stopped this project. 
Maybe I will give it another try.
Your hint about the X11 config is very valuable, I hadn't thought about that.
Thank you for replying.