bigtreetech/CB1 #136: Update OS from Debian 11 to Debian 12

ccatlett1984, 2023-08-21:
https://raspi.debian.net/tested-images/

ccatlett1984, 2023-08-21:
> This isn't really an issue, nor did you give any explanation as to why
> this would be beneficial.

Kernel 6.1 LTS
Support for RUST Programming Language

Multi-Generational Least-Recently-Used (aka MG-LRU; though this not yet
enabled by default). To quote in-kernel documentation, this memory-minded
feature: “…optimizes page reclaim and improves performance under memory
pressure”

6.1 includes a “bunch of performance improvements” to the Btrfs file
system, including a new block group tree to speed up mount on large
filesystems, additional io_uring integration, tweaked sysfs exports; and
“outstanding FIEMAP speed improvement”.

EXT4 filesystem benefits from a flurry of fixes, cleanups, and tuneups, the
latter including no longer trying to prefetch block allocation bitmaps for
read-only file systems.

Other changes:

Kernel Memory Sanitizer (KMSAN) merged
More LoongArch CPU support
Kernel can decompress + launch in architecture-agnostic way on EFI systems
Faster Intel Memory error decoding via EDAC driver
Maple Tree data structure support
New security controls on the ability to create user namespaces
Kernel will print CPU core where segmentation fault occurs

freakydude, 2023-10-21:
I think the point is, is there the will to upgrade to newer debian / kernel versions at all or do we stuck with debian bullseye on cb1 forever?

DerDreschner, 2024-02-11:
FYI: The CB1 is supported by Armbian. This build is maintained by BigTreeTech and is based on Debian 12. You just need to install Klipper manually via KIAUH.

You can find more information under https://www.armbian.com/bigtreetech-cb1/

freakydude, 2024-02-12:
@DerDreschner Thx for this information. 

best regards

RoisinDubh79, 2024-02-22:
> FYI: The CB1 is supported by Armbian. This build is maintained by BigTreeTech and is based on Debian 12. You just need to install Klipper manually via KIAUH if you need it. As it's maintained by BTT directly, I would consider this repository here abandoned.
> 
> You can find more information under https://www.armbian.com/bigtreetech-cb1/

BTT Pad 7 specific drivers, overlays, and scripts are missing from the armbian CB1 images, so it's unfortunately not a "drop in replacement" for the previously provided CB1 images.