bigtreetech/CB1 #137: Cannot use sunxi-fel to connect to CB1 eMMC

vader7071, 2023-08-21:
**Setup:**
- Windows 10
- CB1 eMMC v1.1
- Downloaded sunxi-fel.exe on 19 Aug 2023 from this GitHub (along with all other necessary files).

**Issue:**
I connect my BigTreeTech CB1 via my BigTreeTech PI4B adapter to my computer via USB cable. Jumpers are set as per instructions. I run Zadig and find the "unknown device" (yes, the USB ID matches) and click "Install Driver". Zadig completes.

Next I go to Powershell and run the command `.\sunxi-fel.exe -v ver` and I get the following error:

```
Program 'sunxi-fel.exe' failed to run: The specified executable is not a valid application for this OS platform.At
line:1 char:1
+ .\sunxi-fel.exe -v ver
+ ~~~~~~~~~~~~~~~~~~~~~~.
At line:1 char:1
+ .\sunxi-fel.exe -v ver
+ ~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ResourceUnavailable: (:) [], ApplicationFailedException
    + FullyQualifiedErrorId : NativeCommandFailed
```

Am I downloading the wrong file?

GePo00, 2023-09-24:
I could run this pgm. Maybe you have not the correct version.
But running this pgm it ends successfully. The verify says the new driver is installed correctly.
But windows still says “unknown device #1” !

vader7071, 2023-09-29:
@GePo00 The ZaDig program works fine (the USB driver program). The issue I am having is with sunxi-fel program to write the uboot.

@NaMoUrGCh yes, I have tried running sunxi-fel from multiple directories, thinking I might not have been in the correct folder. I even went as far as to specify the exact location of sunxi-fel when I executed the command via PowerShell. To try and ensure this wasn't the issue, I created a folder on my C drive labeled "c:\cb1" and tried running everything from that location to ensure I had my path as simple and direct as possible. Still did not work.

MrG1981, 2024-03-28:
Same issue here :(