bigtreetech/CB1 #139: CB1/RPI4

HashTag81, 2023-08-26:
Hi Team,
	
	I seem to be having trouble with my CB1 V1 not wanting to boot after I’ve shut it down via Mainsail. I’m had to buy another cb1/rpi4 as i thought it might have been damaged and now I’m having the same issues. printing for days and then nothing. i refreshed the board with all the new updates, shut it down and now nothing again! i do have a Manta M8P board waiting for my Zero G one.1 hydra build and I’m concerned that ill be having the same issues. 

Can you please explain why I’m having these issues?

420noscope-exe, 2023-10-16:
linking this to issue #125.  Same thing happened to me.  My printer stays online for a month, then suddenly stops working, requiring the SD card to be re-imaged. @bigtreetech has not responded yet.

Abyss8, 2023-10-19:
I have problem to connect browser and putty, and when i go to wifi settings on Pad 7 it freezes.

closed 2024-01-27 by HashTag81