bigtreetech/CB1 #140: CB1 internal temperature monitoring?

Valiance, 2023-08-28:
The CB1 unit burns hot at times.   Is there a safety internal temperature monitor for max temp shut down?   Is there a more civil internal temperature reporting it does to the OS, or say... Klipper?  Like a temp sensor I could use to trigger a fan2 on/off?  This could help prevent forest fires...    If not, perhaps getting an official statement from BTT in the form of a warning: 'There is not an internal temperature monitoring system available for the CB1'  Granted, I know that is a bit much, but still... asking.

Pneumanifest, 2023-08-28:
This will get you the temperature in klipper,

``` [temperature_sensor CB1]
sensor_type: temperature_host
 ```
Or install ls-sensors terminal,
```
sudo apt install ls-sensors
``` 
When that is done run command,
```
sensors 
```

Valiance, 2023-08-28:
> And this is an example of how to control fins with the built-in
> temperature sensor.
>
> [temperature_fan Case_fans]
> pin: PD4
> sensor_type: temperature_host
> off_below: 0.4
> min_temp: 10
> max_temp: 90
> target_temp: 55
> control: pid
> pid_Kp: 2
> pid_Ki: 4
> pid_Kd: 0.1

Thank you so much, Patrick.    This is beautiful.

420noscope-exe, 2023-10-16:
@Valiance make sure to close you issue if it was answered

Valiance, 2023-10-17:
> @Valiance <https://github.com/Valiance> make sure to close your issue if
> it was answered

Yes, it is working.
"Reality is the Ultimate Role-playing game." -- Victor A. Waters '88

closed 2023-10-17 by Valiance