bigtreetech/CB1 #153: 32-bit image

airbornetrooper82573, 2023-11-11:
I have issues with CANbus and even after going from v1 board to v2 and now using CANbridge mode it's still happening. From research, it looks like 32-bit Bullseye is solving many of the issues. Is there a 32-bit image available for the CB1?

sntlewis, 2024-03-10:
reviving a dead thread?  The change of the TRSYNC_TIMEOUT doesn't fix the problem completely.  I still get the occasional timeout issue.  So any chances of BTT adding a 32-Bit Bullseye/Bookworm image?

airbornetrooper82573, 2024-03-10:
> reviving a dead thread? The change of the TRSYNC_TIMEOUT doesn't fix the problem completely. I still get the occasional timeout issue. So any chances of BTT adding a 32-Bit Bullseye/Bookworm image?

I know some recent changes with Klipper is supposed to help but I am still using the MCU modification and it works for me.

sntlewis, 2024-03-10:
I just did a klipper update today.  I generally give it couple of tries before jumping into updating the time out value.

viniavila, 2024-09-04:
@airbornetrooper82573 
I'm facing the same issue, TRSYNC_TIMEOUT helps. I'm using danger-klipper, which allows modify this parameter from printer.cfg.
```[danger_options]
multi_mcu_trsync_timeout: 0.075```
But this parameter doesn't solve all the problems for me. I reduced my extruder microsteps to only 8 (stealthburner stepper is a 0.9 degree so `full_steps_per_rotation: 400`). Reducing print speed is essential, if I force 300 mm/s and more with detailed pieces, I falls into Timer Too Close error. Using 150 or 200 mm/s, I don't have any issues. Turning off Input Shaping also eliminates this issue.

I don't have any problem compiling the OS image myself, but without clear instructions on how to target 32bit version, can be very painful.

igeighty, 2025-01-07:
i too would like info on compiling a 32bit image to combat these timer errors. it has been a successful fix for pi users, and rather than move to CM4 i would like to be able to compile an image for CB1 and CB2. could anyone point some hints ?

frutabruta, 2025-01-07:
I use Manta M8P + CB2 + U2C + EBB42. I had issues with timeouts. I solved the issue, which started with upgrade to CB2, with this guide: [BigTreeTech U2C](https://github.com/docgalaxyblock/KlipperMisc/blob/main/CAN/BigTreeTech-U2C/README.md)
The main change was NOT to use the candlelight firmware, but use the U2C as Klipper bridge ( _Klipper USB-to-CAN adapter (advanced way)_) section. And I also changed the CAN network interface setting from 1024 to 128, again, according to the guide. The only difference is that i use 1000000 bitrate, instead of 500000 used in the guide.
```
auto can0
iface can0 can static
    bitrate 1000000
    up ifconfig $IFACE txqueuelen 128
```

igeighty, 2025-01-07:
thanks frutabruta
i upgraded from cb1 to cb2, only to have more errors, i have completely ruled out cabling, i use manta 8p cb2 and sbb2209 connecting using canbus supplied by the manta. changing to u2c isnt really an option here, i will just purchase CM4 and run 32bit that way.. meanwhile searching on how to compile 32bit for btt cb1 and cb2