bigtreetech/CB1 #158: Issue with system time

morrmaker, 2024-01-06:
I have reimaged my board several times and it always sets the back to July 16, 2023.  I'm able to manually set it to current date time with timedatectl however, after having printer turned of for awhile I now notice it is 4 hours behind.  Is there a way to set NTP service to keep this in sync.  Timedatectl shows "NTP servie: n/a"

Chinstrap777, 2024-01-06:
change your timezone in "system.cfg" to "America/Seattle" or whatever your designator is. I used google to find mine.

morrmaker, 2024-01-07:
Yes. Forgot to mention I already set the timezone.

freegithubuser, 2024-01-30:
I also have the same problem.
Have set the timezone in the system.cfg as well as at the command line and I still get the same error.

```bash
#!/bin/bash

biqu@anet:~$ timedatectl status
               Local time: Tue 2024-01-30 15:44:44 AEDT
           Universal time: Tue 2024-01-30 04:44:44 UTC
                 RTC time: Fri 1970-01-02 00:21:24
                Time zone: Australia/Melbourne (AEDT, +1100)
System clock synchronized: no
              NTP service: n/a
          RTC in local TZ: no

Sayon13, 2024-03-13:
Same problem here, I cannot update via Mainsail because the system date is so old. Even in Kiauh I get the "E: Release file for http://deb.debian.org/debian-security/dists/bullseye-security/InRelease is not valid yet (invalid for another 243d 8h 37min 8s). Updates for this repository will not be applied."

Sayon13, 2024-03-13:
[image]

Sayon13, 2024-03-13:
[image]
[moonraker.log](https://github.com/bigtreetech/CB1/files/14591917/moonraker.log)

freegithubuser, 2024-03-13:
The problem appears that NTP which is responsible for network time isn't running/working.
I have tried changing NTP servers and configuration but still had no luck. I'm sure someone has figured it out, but until then I have found that I need to SSH into it and set the date and time manually after each boot for it to stay current. Some short power off states (a few minutes) or while whilst rebooting seem to freeze for that short period,

`sudo date "YYYY-MM-DD HH:MM:SS"`

Sayon13, 2024-03-13:
The think that kills me inside is that it was working fine before I update it.. I gave up on it and just change it for a rpi 3b+

Chinstrap777, 2024-03-13:
> I also have the same problem. Have set the timezone in the system.cfg as well as at the command line and I still get the same error. All OS updates are installed.
> 
> ```shell
> #!/bin/bash
> 
> biqu@anet:~$ timedatectl status
>                Local time: Tue 2024-01-30 15:44:44 AEDT
>            Universal time: Tue 2024-01-30 04:44:44 UTC
>                  RTC time: Fri 1970-01-02 00:21:24
>                 Time zone: Australia/Melbourne (AEDT, +1100)
> System clock synchronized: no
>               NTP service: n/a
>           RTC in local TZ: no
> 
> biqu@anet:~$ ntpq -p
>      remote           refid      st t when poll reach   delay   offset  jitter
> ==============================================================================
>  used-to-be-ntp. .INIT.          16 u    - 1024    0    0.000   +0.000   0.000
> ```

Looks like you have to: 

sudo debian-config

To actually change the time zone and time

PH1980, 2024-03-15:
Hello there!

I solved the problem. Had the same issue.

got to 

https://wiki.debian.org/DateTime

and look for installation of ntp. Do it. Nano into the mentioned file and add local servers from the local sources under "Managing NTP Pools" right under the ones that are mentioned in the file. Save the file and reboot.

Just did it and it works now for me.

freegithubuser, 2024-03-15:
> Hello there!
> 
> I solved the problem. Had the same issue.
> 
> got to
> 
> https://wiki.debian.org/DateTime
> 
> and look for installation of ntp. Do it. Nano into the mentioned file and add local servers from the local sources under "Managing NTP Pools" right under the ones that are mentioned in the file. Save the file and reboot.
> 
> Just did it and it works now for me. Did Update and Upgrade and now downloaded KIAUH and install Klipper.

Thanks PH1980, worked a treat!