bigtreetech/CB1 #159: upgrade the CB1 image to Debian 12 Bookworm

Chinstrap777, 2024-01-06:
Upgrade the CB1 image to Debian 12 Bookwoom to allow use of /dev/serial/by-id/

frutabruta, 2024-01-06:
The `ls /dev/serial/by-id/` works for me in the V2.3.2 minimal image.
```console
biqu@BTT-CB1:~$ ls /dev/serial/by-id
usb-Klipper_stm32g0b1xx_4A000A000F504B4633373520-if00
```
It detected Manta M8P properly. What kind of board are you trying to detect?

Chinstrap777, 2024-01-10:
BTT Pad 7 using the newest (JUL2023) full image of the CB1.

clevelandcs, 2024-01-23:
@Chinstrap777 I tested upgrading to Bookworm after install which worked (although I had to reinstall Klipper and Moonraker with Kiauh afterwords). Another alternative I'm testing out now is the Armbian image which is built with kernel 6.1 LTS and Debian 12 out of the box. 
https://www.armbian.com/bigtreetech-cb1/
https://docs.armbian.com/User-Guide_Getting-Started/
I'm not sure if BTT will be updating this image again considering the status of this repository. The Armbian image is a lot newer dated November 30th and 6.1 is an LTS kernel version unlike 5.16 as ships on the CB1 image in this repository.

Chinstrap777, 2024-01-25:
> @Chinstrap777 I tested upgrading to Bookworm after install which worked (although I had to reinstall Klipper and Moonraker with Kiauh afterwords). Another alternative I'm testing out now is the [...]

This has worked beautifully so far. I loaded a BTT Pi v1.2 with the CLI version and KIAUH and haven’t had any problems. Granted, I haven’t tried using the CAN or Screen ports.

clevelandcs, 2024-01-25:
CAN should be working fine at least in the context of the OS. Firmware and configuration of the CAN boards seems to always be a bit finicky at least for me but that hasn't had anything to do with OS images for me. I can't provide any insight into the display outputs as I run all my printers headless and use a single Pi 3B+ with touchscreen and Klipperscreen mounted to the wall along with Mainsail to manage all my printers. That said I'd doubt there would be an issue with HDMI but iirc the BTT boards haven't really supported the other display out modes.

DerDreschner, 2024-02-11:
FYI: The CB1 is supported by Armbian. This build is maintained by BigTreeTech and is based on Debian 12. You just need to install Klipper manually via KIAUH.

You can find more information under https://www.armbian.com/bigtreetech-cb1/

RoisinDubh79, 2024-02-21:
> This has worked beautifully so far. I loaded a BTT Pi v1.2 with the CLI version and KIAUH and haven’t had any problems. Granted, I haven’t tried using the CAN or Screen ports.

Do your status lights on the front of the BTT Pad 7 work after upgrading to the Armbian CB1 image?
The original CB1 image has some special scripts and binaries for dealing with the status lights and some other things.

dekay, 2024-03-10:
> FYI: The CB1 is supported by Armbian. This build is maintained by BigTreeTech and is based on Debian 12. You just need to install Klipper manually via KIAUH if you need it. As it's maintained by BTT directly, I would consider this repository here abandoned.

There have been a number of commits even within the last few days to the CB1 kernel by BTT.  I would not call it abandoned.

https://github.com/bigtreetech/CB1-Kernel/commits/kernel-5.16/

Chinstrap777, 2024-04-10:
Until 3 weeks ago, there hadn’t been an updated CB1 image since July 2023. For a very popular software/hardware product that is very odd to not update the image after 8 months, but to update another version (Armbian) every few weeks is odd still. I only stumbled upon the Armbian image because another user here commented on my issue request. BTT would help users a lot more by just mentioning that it exists in the user manual and GitHub readme for the CB1. Just my two cents.

Chinstrap777, 2024-04-10:
> Do your status lights on the front of the BTT Pad 7 work after upgrading to the Armbian CB1 image? The original CB1 image has some special scripts and binaries for dealing with the status lights and [...]

And I’m using a Manta M5P now.

viniavila, 2024-09-12:
> @Chinstrap777 I tested upgrading to Bookworm after install which worked (although I had to reinstall Klipper and Moonraker with Kiauh afterwords). Another alternative I'm testing out now is the [...]

How to make wifi work? I tried this, but I can only ssh into CB1 using ethernet cable. Inside there, I tried to setup Netplan to recognize my wireless network through wlan0, based on https://docs.armbian.com/User-Guide_Networking/, but no success. Sometimes it connects, and sometimes no. When it connects, I can't ssh into them (checked the right ip using my network mobile app)...

Townie-au, 2024-09-14:
> And I’m using a Manta M5P now.

Hey

How did you get the wifi working?
Missing a driver?

Townie-au, 2024-09-16:
How do I connect to the terminal beforehand?

Pneumanifest, 2024-09-17:
> How did you get the wifi working? Missing a driver?

No drivers or other installs needed. You enter the wifi information in the `boot/system.cfg` file as described here. https://github.com/bigtreetech/CB1#wifi-settings

Schmuecki, 2024-09-17:
> Must install network - manager (sudo apt install network-manager) after that you can config wlan via armbian-config

This applies to the Armbian image (Kernel 6.4, Debian 12.7).
It is easier with the original image from Bigtreetech (Kernel 5.16, Debian11) simply edit /boot/system.cfg.

Townie-au, 2024-09-17:
> Must install network - manager (sudo apt install network-manager) after that you can config wlan via armbian-config

Thanks mate. The new image has network-manager already and ethernet works.

Townie-au, 2024-09-17:
> > How did you get the wifi working? Missing a driver?
> 
> No drivers or other installs needed. You enter the wifi information in the `/boot/system.cfg` file as described here. https://github.com/bigtreetech/CB1#wifi-settings

Yes. This is with debian.