bigtreetech/CB1 #160: Getting MCU Protocol error in Klipper

Windbane, 2024-01-19:
Klipper reports: ERROR

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-85-gd785b396-dirty
MCU(s) which should be updated:
mcu: Current version v0.12.0-85-gd785b396
CB1: Current version v0.11.0-241-gffb5105b
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'CB1': Command format mismatch: query_adxl345 oid=%c rest_ticks=%u vs query_adxl345 oid=%c clock=%u rest_ticks=%u

Updated the firmware for my SKR3 board, but how do I update the MCU on the CB1? I also commented out the [adxl345] and [resonance_tester] sections in my printer.cfg file and klipper will run, but if I need to redo my input shaping I would need to update the CB1 mcu...

RandomWon, 2024-01-20:
I also now have this problem during customizing my Voron and after a update. I am attempting to research it.

Pneumanifest, 2024-01-20:
Re-run the make for Linux process, https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code

Windbane, 2024-01-20:
@Pneumanifest  thanks this worked.

closed 2024-01-20 by Windbane

Titus-A-Duxass, 2024-01-21:
I am experiencing the same issue.
Unfortunately I'm running BTT CB1 and the instructions are not working for me.

Pneumanifest, 2024-01-21:
@Titus-A-Duxass it's only if the host MCU firmware is out of date.  If it's the regular MCU that is out of date you need to make and flash that according to the controller board's requirements.

Titus-A-Duxass, 2024-01-21:
> @Titus-A-Duxass it's only if the host MCU firmware is out of date. If it's the regular MCU that is out of date you need to make and flash that according to the controller board's requirements.

Thanks for the response. I'll try updating the regular (board?) MCU.

Pneumanifest, 2024-01-21:
> Thanks for the response. I'll try updating the regular (board?) MCU. 
Yes the controller board. 
What does the error message say? It will tell you which one is out of date.

Titus-A-Duxass, 2024-01-21:
> > What does the error message say? It will tell you which one is out of date.

Your Klipper version is: v0.12.0-85-gd785b396
MCU(s) which should be updated:
mcu: Current version v0.11.0-122-ge6ef48cd
EBBCan: Current version v0.11.0-122-ge6ef48cd
display: Current version v0.12.0-60-g0665dc89
Up-to-date MCU(s):
<none>

frutabruta, 2024-01-21:
Klipper version in CB1 and in the printer board (for example Manta, SKR3 etc.) have to be in sync. So don't download a precompiled version from the web. The correct thing is to update the Klipper in CB1 (KIAUH is good for that), and then on the CB1 compile firmware for SKR/Manta and then upload it. This will create matching versions in both CB1 and printer board processor.

Pneumanifest, 2024-01-21:
> > > What does the error message say? It will tell you which one is out of date.
> 
> Your Klipper version is: v0.12.0-85-gd785b396 MCU(s) which should be updated: mcu: Current version v0.11.0-122-ge6ef48cd EBBCan: Current version v0.11.0-122-ge6ef48cd display: Current version v0.12.0-60-g0665dc89 Up-to-date MCU(s):
> 
> Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software.
> 
> mcu 'EBBCan': Command format mismatch: query_adxl345 oid=%c rest_ticks=%u vs query_adxl345 oid=%c clock=%u rest_ticks=%u

Yeah you don't even have the cb1 as MCU in that list. You need to update the firmware on the controller board and the ebb. The link I gave is not for those as the OP was looking how to update the host as MCU part.

Titus-A-Duxass, 2024-01-21:
> Klipper version in CB1 and in the printer board (for example Manta, SKR3 etc.) have to be in sync. 
 
That I understand.

So don't download a precompiled version from the web. 

A precompiled what? Klipper or the CB1 image?

The correct thing is to update the Klipper in CB1 (KIAUH is good for that), and then on the CB1 compile firmware for SKR/Manta and then upload it. 

Okay I've nearly got this - Update klipper on the CB1 with KIAUH - understood.
How do I compile the firmware on the CB1, is this via the menuconfig?

It'll all worked fine on my V2.4 M8P and did work fine on the M5P - now I get these issues.
This will create matching versions in both CB1 and printer board processor.

Titus-A-Duxass, 2024-01-21:
> Yeah you don't even have the cb1 as MCU in that list. You need to update the firmware on the controller board and the ebb. The link I gave is not for those as the OP was looking how to update the host as MCU part.

I've just flashed both the M5P and the EBB36 and got UUIDs for both.

Pneumanifest, 2024-01-21:
> I've flashed both the M5P and the EBB36 and got UUIDs for both before I got this MCU protocol error.

You need to use make menuconfig in the klipper directory on the cb1 to make new firmware for the M5P and ebb. Then install them. They show that the firmware installed is 0.11 and it needs to be the new 0.12 that you updated to.

Titus-A-Duxass, 2024-01-21:
> You need to use make menuconfig in the klipper directory on the cb1 to make new firmware for the M5P and ebb. Then install them. They show that the firmware installed is 0.11 and it needs to be the [...]

Thanks, I'll run through the whole process again. I have just updated the firmware via make flash FLASH_DEVICE=0483:df11 (with the M5P in DFU mode) but it did return an error.

Titus-A-Duxass, 2024-01-21:
And it continues....
I configure katapult and klipper for (this is where I'm confused - M5P or CB1?), flash it with the M5P/CB1 in DFU mode, and then run ~/katapult/scripts/flash_can.py -i can0 -q and I get the UUID for the MCU (M5P or CB1?).

Then I configure katapult and klipper again, then put the EBB36 in to DFU mode and flash katapult.

~/katapult/scripts/flash_can.py -i can0 -q then gives me both UUIDs. 
Power off and reconnect the EBB36 via CAN, check that the network interfaces is as it should be.
Now ~/katapult/scripts/flash_can.py -i can0 -q just shows me the UUID of the EBB36.

Something is not working/staying with the M5P configuration.

Pneumanifest, 2024-01-21:
> I have just updated the firmware via make flash FLASH_DEVICE=0483:df11 (with the M5P in DFU mode) but it did return an error.

You have to do a new ```make menuconfig``` command and then ```make clean``` then ```make``` to actually compile the new firmware first.

frutabruta, 2024-01-21:
‘~/katapult/scripts/flash_can.py -i can0'
The M5P vanished, because I got successfully connected to the Klipper. The EBB42 is the one, which did not connect to the Klipper.

Titus-A-Duxass, 2024-01-21:
> > I have just updated the firmware via make flash FLASH_DEVICE=0483:df11 (with the M5P in DFU mode) but it did return an error.
> 
> You have to do a new `make menuconfig` command and then `make clean` then `make` to actually compile the new firmware first.

I did that on my 2nd attempt. I couldn't get the firmware to flash over the SD card, so I uploaded the firmware.bin to /klipper/out and flash it via dfu-util - it returned no errors and said that it was successful.

Pneumanifest, 2024-01-21:
If you are using katapult then you should be updating via the katapult script and not the make flash command. And you have to tell it which UUID to flash to. 
If the M5P gets connected to klipper then it won't show the id anymore same when the ebb connects.

Titus-A-Duxass, 2024-01-21:
Fixed, I ran the ~/katapult/scripts/flash_can.py -u UUID again and restarted and it appears to be working again.

I now have:
mcu (stm32g0b1xx) Version v0.12.0-85-gd785b396
mcu EBBCan Version v0.12.0-85-gd785b396
mcu display Version v0.12.0-85-g0665dc89

Most of the errors are probably down to finger trouble or not reading things correctly.

Thank you all for your assistance, it is greatly appreciated. At 62 yo I never expected to be doing this sort of thing when I got into 3D printing - it keeps the old grey matter active.

Danke
TAD

Pneumanifest, 2024-01-21:
Sure thing, glad to see you got it. There's always some fun challenge waiting around the corner. I like it that way it keeps things from going stale.

RandomWon, 2024-01-21:
Ive been using klipper for a couple years now but im still a novice.  I did rebuild the firmware. I was in a hurry to just get things working so I didnt check but I suspect there was no change to that firmware since I just bought this SKR3 two or three weeks ago and flashed it.  This is a new build and I have been loading my machine up with all the latest bells and whistles and at times I have broken things and had to take a step back. Im will propably try a fresh build on the PI at this point.

Will-wastelander, 2024-01-21:
I have this on my Voron v2.4, recompiled rpi mcu and seemed to fix it.

I-M-K4tana, 2024-01-23:
I am having a similar issue 

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-88-g2e8b54ae
MCU(s) which should be updated:
mcu: Current version v0.12.0-85-gd785b396
rpi: Current version v0.11.0-122-ge6ef48cd
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'rpi': Command format mismatch: query_adxl345 oid=%c rest_ticks=%u vs query_adxl345 oid=%c clock=%u rest_ticks=%u

I have a ender 3 s1 running klipper whit a btt cb1

Pneumanifest, 2024-01-23:
> Fix by recompiling and flashing the firmware.
> 
> Your Klipper version is: v0.12.0-88-g2e8b54ae MCU(s) which should be updated: 
> rpi: Current version v0.11.0-122-ge6ef48cd Up-to-date MCU(s):

Do as it says and update your firmware. 

To compile the Klipper micro-controller code, start by configuring it for the "Linux process":
 `cd ~/klipper/
`
`make menuconfig
`
In the menu, set "Microcontroller Architecture" to "Linux process," then save and exit.
To build and install the new micro-controller code, run:
`sudo service klipper stop
`
`make flash
`
`sudo service klipper start
`

link to klipper docs: https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code

I-M-K4tana, 2024-01-23:
> Do as it says and update your firmware.
>
> To compile the Klipper micro-controller code, start by configuring it for the "Linux process": `cd ~/klipper/ ` `make menuconfig ` In the menu, set [...]

Thank you very mutch this fixed it I was going crazy

Explore2Create, 2025-06-07:
Can someone explain this as if I were your grandmother... I am so lost.