bigtreetech/CB1 #167: Ghost touches with CB1 and BTT TFT 35 SPI

designnc, 2024-02-24:
It is somewhat annoying, but it is almost always difficult to press the elements or there are random ghost touches

OstroMa, 2024-02-29:
@designnc 
when BigTreetech compiled a new Image, then is thie fix included.

Or you can build it self 
https://github.com/bigtreetech/CB1-Kernel

gneu42, 2024-03-25:
I also have ghost touches. Eventhough I have reinstall everything with the latest 2.3.4 minimal image https://github.com/bigtreetech/CB1/releases/download/V2.3.4/CB1_Debian11_minimal_kernel5.16_20240319.img.xz
That image does not fix the ghost touches, it just change the buttons that are ghostly touched.
Now it is often the temerature  and then the ABS preset that are used by the ghost.
Yesterday after starting the printer and left it idle, when I came back to it after 30 minutes, the bed was at 90°C and the hotend at 220°C. 
This is a dangerous situation.
What can be done to solve this beside replacing the BTT PI  and TFT35 SPI by a genuine RPI PI and a DSI display. Which never gave me such problems.

oboland, 2024-06-18:
Same things happen to me with the latest 2.3.4 version that is supposed to fix it. Hope BTT fix it soon...

gneu42, 2024-06-18:
I fix it definitively.
I trash the BTT PI and SPI display and installed a Raspberry PI 4 and DSI display. All my problems are gone now.
Never go back to BBT PI