bigtreetech/CB1 #168: The /tmp mount point is too small

iandouglas, 2024-03-03:
I bought a Troodon 2.0 Pro, which includes the CB1 board and the OS installed is "BTT-CB1 2.3.3 Bullseye".

The /tmp folder that Formbot sets up is only 512MB which is too small for really big prints. I commented out the /tmp mount point in /etc/fstab, but after rebooting it appears that /tmp is mounted from /dev/zram2 which is even smaller at 478MB.

What's the best way to NOT mount a /tmp device and let /tmp just live on /dev/mmcblk1p2 ?

Erosion139, 2024-10-23:
Bumping this because I also run into this same issue with gcode over around 400mb. This TMP is also cached in RAM which makes it a pain to relocate. I havent been able to do it. Might just buy a CM4 and make my own image.

iandouglas, 2024-10-23:
I was able to do it by removing the tmp mounting at startup so it just uses the filesystem as is

Erosion139, 2024-10-23:
> I was able to do it by removing the tmp mounting at startup so it just uses the filesystem as is

But will that resolve the TMP ram cache issue? I did attempt some kind of TMP workaround but it ended up breaking some things. I don't know how to change things like that in a linux environment, could you point me to a resource?