bigtreetech/CB1 #173: SSH connection is impossible even visible on wifi

JEOBDX, 2024-04-01:
Hello everyone,

When I turned my 3d printer back on, I can no longer connect to klipper.

my BTT Pi v1.2 is visible on the wifi network but I can no longer connect.

I reflashed viaUsing balenaEtcher with the latest version v2.3.4 and put the wifi parameters in the config.cfg file but without success.

Do you have any ideas ?

thank you in advance for your help

manotroll, 2024-04-27:
inside the system.cfg file you must uncomment
#hostname="BTT-CB1" and what is inside the "" can be changed, do not use special characters
and put the wifi password in its normal place
You have to see what username/password you are using for the image

JonLaRue, 2024-05-01:
Hello I have had this issue as well. I have found for whatever reason, power cycling the BTT Pi over and over can help. I can currently connect via IP to mainsail with my phone.. but for whatever reason I cannot with my PC... I am using the same bookmark. I am also unable to SSH into the Pi...

manotroll, 2024-05-02:
Check if network isolation is not enabled on the router

JonLaRue, 2024-05-02:
Changing AP Isolated from "No" to "Yes" did not fix.

manotroll, 2024-05-02:
then return to No
But can you ping? What program are you using to make the ssh connection?

JonLaRue, 2024-05-04:
Factory Resetting Wifi Network and then manually selecting IP address has solved the issue.

JEOBDX, 2024-05-04:
HI ALL,

What ever reason , I flash again and succed to connect ssh after 
"uncomment #hostname="BTT-CB1" and what is inside the "" can be changed, do not use special characters and put the wifi password in its normal place " in the CFG file as @manotroll said in comment before.

I continue to work with and install Klipper to see if everything all right ....