bigtreetech/CB1 #174: Unnamed GPIO with gpiod on CB1

stantonik, 2024-04-02:
I want to control GPIO on the CB1. I saw that it is possible with [gpiod](https://wiki.radxa.com/Gpiod). In my case, pin names don't appear but there is more than 300 pin and without the alias I can't know what is who. Do you know how to solve it ? [I saw that I need to make a .dts file in the root](https://raspberrypi.stackexchange.com/questions/121655/why-is-gpioinfo-not-showing-me-pin-names) but where can I find which line of a "$sudo gpioinfo" corresponds to which GPIO ?

BadFlamingo666, 2024-05-05:
I am also having this problem. I have applied the "light" overlay in BoardEnv.txt but that seems not to work.

sergei-iliev, 2024-05-23:
Here is my finding on 'gpiochip0' up until now
Header pin 7 ,PC7, corresponds to line 71
Header pin 37 ,PC15 , corresponds to line 79
Header pin 40 , PC8 , corresponds to line 72
I can toggle those header pins using libgpiod C/C++ lib

Pneumanifest, 2024-05-27:
The user manual has the information about the gpio numbering, but here's a screenshot to help.
[image]