bigtreetech/CB1 #179: manta mp8 klipper firmware update

manotroll, 2024-04-27:
When running make menuconfig from klipper, what options do I have to select to update but use the default card?
https://imgur.com/gcr1KSL
This part is different from the manual

and now the command ls /dev/serial/by-id/*
reports that there is no serial port

biqu@BTT-CB1:~$ ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory
biqu@BTT-CB1:~$
biqu@BTT-CB1:~$ sudo ls /dev/serial/by-id/*
ls: cannot access '/dev/serial/by-id/*': No such file or directory
biqu@BTT-CB1:~$

shawnchr, 2024-04-27:
Please explain you issue more clearly.
You are all over the place you have a M8P what version is the CB1 mounted onboard?
You have 2 sdcard slots one will work for the CB1 and the works for updating the board firmware.
you will only see output from the ls /dev/serial/by-id/ after a successful firmware install

manotroll, 2024-04-27:
M8P V2.0
yes, the CB1 is plugged in/connected to the M8P
I tried to update via micro SD and renamed the file as dis in the tutorial but the command still doesn't work

shawnchr, 2024-04-27:
Ok Did the file on the sdcard change from firmware.bin to firmware.CUR? check by plugging back into your computer.

Which SDCARD slot are you using the one under the bord where the CB1/CM4 attaches is for those boards.
You need to use the sdcard slot on the side where the USB ports are.

manotroll, 2024-04-27:
yes it changed to firmware.CUR after doing the power on procedure holding reset the led blinked a few times

manotroll, 2024-04-28:
I can record but nothing changes

 sudo dfu-util -d ,0483:df11 -a 0 --dfuse-address 0x8000000:force:mass-erase -D ./klipper.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Performing mass erase, this can take a moment
Downloading to address = 0x08000000, size = 29476
Download        [=========================] 100%        29476 bytes
Download done.
File downloaded successfully

manotroll, 2024-04-28:
I have the same similar problem
It even records but even restarting it doesn't exit UFD mode

manotroll, 2024-04-28:
I found the solution to the problem in a video on YouTube, I'll put it here so you can see
There is an error in the PDF explanation in the manual on how to do it
https://www.youtube.com/watch?v=k8DmTrY2GgY