bigtreetech/CB1 #183: M8P-CB1 trying to add BTT HDMI5

Vinnnocc, 2024-05-14:
Hey all 
I have an M8P v1.1 with a CB1 
I have klipper running along with an EBB36 it’s a fully functional printer. I used Kiauh to install everything.  I’ve decided i wanted to add a touchscreen. So figuring I’d stay with BTT I went and got the HDMI5.   

I went to install Klipperscreen saw Kiauh was the easiest way. So I ssh in and saw it was already installed.  I pulled the sd card and edited the txt file to change to hdmi. Put back into m8p. Fired up printer. Added Klipperscreen to moonracker, added the trusted cert in as well.   Rebooted everything. Still just a blank/black screen.  I plugged the hdmi into port 1 and 0 no luck. I plugged the usbc into m8p usb and walkout no luck. 
I then plugged hdmi into my Mac. With usbc into Mac as well. And it worked can view my Mac screen on the hdmi5. So I know the screen isn’t broken.  

Then i figured I’d remove Klipperscreen with Kiauh. Then i reinstalled with Kiauh.   I did get a question I’m not sure about. Asked if I want to standalone y/n? I selected Y   (Not sure if supposed to). Then it asked about something else but I don’t remember what it said but answers were x and I think w.  I selected X then I did a bunch of stuff (once again not sure if that’s correct)
I can’t find anywhere videos/ instructions that someone else was asked those questions. 

What am i doing wrong? 
could it be as simple as my M8P-cb1 / EBB are very outdated. Wanna say last time I updated firmware was well over a year ago??

[image]
[image]
[image]
[image]
[image]
[image]

najwaa-debug, 2024-06-04:
Having a similar issue with HDMI 7 screen. Were you able to find a resolution?

Vinnnocc, 2024-06-04:
> Having a similar issue with HDMI 7 screen. Were you able to find a resolution?

Yup it was the same supplied USB micro adapter they sent.

najwaa-debug, 2024-06-04:
> > Having a similar issue with HDMI 7 screen. Were you able to find a resolution?
> 
> Yup it was the same supplied USB micro adapter they sent.

Appreciate the response. I had gone ahead and ordered another cable just in case so works out 👍🏽

Oldmopars, 2024-06-25:
I have the same issue. M8P CB1 HDMI5, no display. I have tested the screen on another printer, it works fine. I swapped several cables, no luck. Pulled cables from a working printer with Octopus and HDMI5, no display. The screen is good, the cables are good, still no display. I've done the Sudo klipperscreen restart, updated everything klipper related, reinstalled klipper screen, still nothing. Any ideas?

Vinnnocc, 2024-06-25:
> I have the same issue. M8P CB1 HDMI5, no display. I have tested the screen on another printer, it works fine. I swapped several cables, no luck. Pulled cables from a working printer with Octopus and [...]

Yup it was the same supplied USB micro adapter they sent. I went with a micro USB to HDMi cable with no adapter

frutabruta, 2024-06-25:
This is crazy, the BTT5 works great for me with Manta M8P and CB1.
I used the minimal image, installed KIAUH and with that Klipper, Moonraker, Mainsail, KlipperScreen. I didn't have to set anything in the BoardEnv.txt.
But I had to do this:

> If you use a minimal version of the image, additional settings are required after installing klipperscreen.
add needs_root_rights=yes to /etc/X11/Xwrapper.config (sudo nano /etc/X11/Xwrapper.config, add needs_root_rights=yes)

Mentioned in the releases of [2.3.2](https://github.com/bigtreetech/CB1/releases/tag/V2.3.2)

Vinnnocc, 2024-06-25:
> This is crazy, the BTT5 works great for me with Manta M8P and CB1. I used the minimal image, installed KIAUH and with that Klipper, Moonraker, Mainsail, KlipperScreen. I didn't have to set anything [...]

Well technically i didn’t have to do anything other then install KlipperScreen with KIAUH. Then add everything to the config.    I followed the basic instructions but my adapter was bad. Used a new cable without the adapter and started to work

Oldmopars, 2024-06-25:
I used the BTT image, then went back and uninstalled Klipperscreen and reinstalled it. I even used a known good cable from my 2.4 with BTT HDMI5 and tried it, no luck. The 2.4's cable is a standard to mini HDMI cable, no adapter. I have 3 of these screens. 2 on Octopus, one on the Manta. The Manta is the only one I can't get working. 
I will try the needs_root_rights=yes.

najwaa-debug, 2024-06-25:
> I used the BTT image, then went back and uninstalled Klipperscreen and reinstalled it. I even used a known good cable from my 2.4 with BTT HDMI5 and tried it, no luck. The 2.4's cable is a standard [...]

> I used the BTT image, then went back and uninstalled Klipperscreen and reinstalled it. I even used a known good cable from my 2.4 with BTT HDMI5 and tried it, no luck. The 2.4's cable is a standard [...]

Which version of the BTT image did you use? I was using an older version and updating to the latest version fixed my issue.