bigtreetech/CB1 #185: BTT HDMI 7 Black screen with BTT Pi 1.2

najwaa-debug, 2024-06-04:
Hi, I am using a BTT Pi v1.2 with Klipper for my 3D printer. It works great. I have been able to use a BTT TFT35 screen with it but I wanted a bigger screen so I got the BTT HDMI 7 screen. 

I have plugged in the USB cord to the board and the screen. I have plugged the HDMI cord to the screen and then to the board using the Adaptor. There is only one display port that it can be plugged into on the display board. 
[image]

The screen stays black from the moment the board is turned on. It does not show the BTT logo or turn white at any point. I'm hoping someone else has run into this issue. Thanks for your time!

camelator, 2024-09-04:
same issue found with BTT-HDMI5

najwaa-debug, 2024-09-04:
> same issue found with BTT-HDMI5

I was able to resolve my issue by using the latest Klipper img for the CB1.