bigtreetech/CB1 #188: The Local time and RTC time are different when using the timedatectl command

kookooqoo, 2024-07-31:
Is there any RTC hardware issue or this is normal behavior?

```
[root@localhost RTC]# timedatectl
               Local time: Wed 2024-07-31 00:20:55 UTC
           Universal time: Wed 2024-07-31 00:20:55 UTC
                 RTC time: Wed 2024-07-31 00:20:56
                Time zone: UTC (UTC, +0000)
System clock synchronized: no
              NTP service: inactive
          RTC in local TZ: no
[root@localhost RTC]# timedatectl
               Local time: Wed 2024-07-31 00:20:57 UTC
           Universal time: Wed 2024-07-31 00:20:57 UTC
                 RTC time: Wed 2024-07-31 00:20:57
                Time zone: UTC (UTC, +0000)
System clock synchronized: no
              NTP service: inactive
          RTC in local TZ: no
[root@localhost RTC]# timedatectl
               Local time: Wed 2024-07-31 00:20:57 UTC
           Universal time: Wed 2024-07-31 00:20:57 UTC
                 RTC time: Wed 2024-07-31 00:20:58
                Time zone: UTC (UTC, +0000)
System clock synchronized: no
              NTP service: inactive
          RTC in local TZ: no
```

ptinoz, 2024-10-11:
Install Chrony and it will sync.

`sudo apt install chrony`

StackingLayers, 2025-07-06:
> Is there any RTC hardware issue or this is normal behavior?
>
> ```
> [root@localhost RTC]# timedatectl
>                Local time: Wed 2024-07-31 00:20:55 UTC
>            Universal time: Wed 2024-07-31 [...]

Your RTC is not set to synchronize. It's working as configured. 
Yes ``` timedatectl set-local-rtc 1 --adjust-system-clock ``` to set RTC to adjust the local time for the system clock.