bigtreetech/CB1 #190: CB1 won't connect to wifi

carlosspicywings, 2024-08-24:
Desperate for some CB1 help on Wifi issues. I got a Fabreeko Salad Fork that came with Manta M5P + CB1 (V2.2) (+ EBB36 CANBUS, not hooked up yet). I've been stuck trying to connect the CB1 to Wifi, I can never find it on my ISP's scanner or angryIP. Here's where I'm at: I have M5P powered off VUSB with a red light on M5P, green light on CB1, and flashing green on M5P by the CB1 wifi antenna. I can't see the BTT-CB1 when scanning wifi so I can't get the IP to ssh into it. I've tried re-flashing the microSD through BalenaEtcher instead of RPI, changing my network SSID to be simpler with no spaces, replacing the stock antenna with a real one, splitting my wifi into separate 2.4 and 5 channels, and scanning through both my ISP and AngryIP. I've checked the wifi info is correct in the system.cfg file many times. An ethernet connection works no problem, but I need to run this off Wifi because I'm not doing that long term... Tried connecting to my phone's hotspot and no luck there either. Any suggestions? Or is the next step to replace the CB1?

djuttien, 2024-12-05:
same issue here,,, .. did you get it fixed or replaced?

carlosspicywings, 2024-12-06:
Unfortunately no luck, I gave up and just stuck an ethernet cable in there. I'm looking into adding a Wifi bridge which Reddit told me would work but haven't gotten there quite yet. Let me know if you find anything helpful!

carlosspicywings, 2025-01-01:
@djuttien I finally got the Wifi bridge and am kicking myself for not doing that sooner. It took about 7 minutes to set up and this guy's video walks you through every step. Would have saved me probably 80 hours of troubleshooting... 
https://www.youtube.com/watch?v=GfQ5LYxjFnI

Deapper, 2025-03-12:
sudo nmcli dev wifi connect <network-ssid> password <network-password>

This saves even after reboot