bigtreetech/CB1 #194: BTT 1.12 Not working at all, pls help

HelioNM, 2024-11-12:
I purchase a new BTT Pi 1.12.

I followed the instructions, burn the iso provided here to an SD, edit the config to give the SSID and password, insert SD at the BTT and.... Nothing happen, no connection.

I tried to even use a open wifi network, 3 diferent router, even I tried to share the wifi of my phone, nothing happen, the BBT not connect so I can acces it, I dont know even if it is working or not.

The green led is ON and still. I tried to power it using USB-C cable and also using a 12V PSU of my printer. Nothing.

I tried to use both images provided, the klipper one and the minimal, I tried ballenaetcher, rufus and the RPI imager... Nothing works

Also I tried the network cable connection however the leds of the ethernet port dont even turn on.

Help pls.

neeekw, 2024-12-18:
I also had issues at first, until I realised that the raspberry imager was applying the preconfig settings I had set up when I configured the pi zero 2w that was running my 3d printer. Once I cleared those settings it was fine.

I powered the pi off my laptop while setting it up initially, and used putty to see what the serial interface was spitting out, thats what helped me find the issue. I also used a version 2 releases behind the current one, but once up and running, it updated with no issues.

The only annoying part I found is the username and password of biqu, and no apparent way of changing it.

Pneumanifest, 2024-12-18:
You change username and password the same as any Linux system 
Username change https://www.naukri.com/code360/library/how-to-change-username-in-linux

Password is just, ```passwd``` and follow the prompts.

neeekw, 2024-12-19:
The password I changed, the issue with the username is that if you change it, the system does not start as there are config files and the like in the user directories that klipper uses.

I found a way around that though on printables, by essentially creating an alias user with the same user id as biqu.

Pneumanifest, 2024-12-19:
> The password I changed, the issue with the username is that if you change it, the system does not start as there are config files and the like in the user directories that klipper uses.
> 

Yes when you change username you must go through and change ownership and update anything that is using the old name. This is how it is. It's not a bug or issue of the system. 

The best way if you must have your own user name,  is to use the minimal image set up things from scratch and use KIAUH to install all the Klipper stuff. Then you'll have a clean system with your username. Instead of using bandaids that might cause issues in the future.  In general though there's no need to change the username, setting a new password is good for added security though.

neeekw, 2024-12-19:
I am aware that that is how linux works, my understanding though is that you have to use the image that BTT provide due to changes in hardware.

I'll give the minimal image a go though, the reason I want to change the username is to that it fits into my "ecosystem" :)

I plan on making a bunch of changes to it anyway, ssl, iptables etc, so I will give the minimal route a go

Cheers

Pneumanifest, 2024-12-19:
There's also an armbian official image for it on the armbian website if you are going the clean install route.