bigtreetech/CB1 #197: no /dev/serial

OldMan2525, 2024-12-26:
There is no /dev/serial directory created. None of the internet posted fixes for Debian Bookworm work.  This is with a new CB1, with fresh installs. I tried both minimal and Klipper full versions of the latest 3.0.0 images  (CB1_Debian12_minimal_kernel6.6_20241219.img and CB1_Debian12_Klipper_kernel6.6_20241219.img)

Very frustrating.

shawnchr, 2024-12-26:
The /dev/serial is only created when a functional serial device is attached a functioning its dynamic. I am using the new image and the old image and they both function correctly.

OldMan2525, 2024-12-27:
> The /dev/serial is only created when a functional serial device is attached and functioning, its dynamic. I am using the new image and the old image and they both function correctly.

Thank you for your quick response!  I figured this out a couple hours ago,  but would have saved myself many more, had I checked back earlier to read your reply!

Esoteric’s guide was what finally led me to success getting the EBB SB2209 RP2040 talking over CAN to the Manta M8P and CB1.  Even so, Esoteric left out that you *must* double click RST before each attempt to flash Klipper, to force the M8P, and later the EBB, into Katapult mode. The rest of his instructions were pretty much spot-on, so that omission in the hand-holding led to some additional head-scratching.

closed 2024-12-27 by OldMan2525