bigtreetech/CB1 #198: "The disk you attached was not readable by this computer" / Debian 12 image / MacOS

bjull001, 2024-12-29:
Hello everyone, 
I have a problem where I can't find a solution. 
I'm trying to flash the new image of the V3.0.0 image on a SD card using Pi Imager or BalenaEtcher, but after everything finishes and I reinsert the SD card, I get the error message mentioned above. 
My process is as follows: 
- Insert SD Card (I tried two SD cards and two SD card readers)
- Erase it using SD card formatter (5.0.3) or Pi imager ( I tried both)
- Downloading the image ( Minimal or Klipper, both finished successfully, and I downloaded them several times. Also tried the armbian build.)
- Use Pi imager (1.8.5) or BalenaEtcher (1.19.25) to flash the image on the SD card  
- Process finishes successfully and ejects the SD card. 
- Reinsert the SD card and get the error message mentioned above. (Options are eject, ignore or initialize. )

I'm using a 2019 Macbook Pro with Sonoma 14.6.1 and the CB1_V2.2 with the adatper board. 

Also flashing the V2.3.4 image works just fine. 

Is there anything I'm missing? 
Thank you in advance for any help!

shawnchr, 2024-12-29:
I have not found an issue other than using a windows computer to edit the file. I have tried everything I am not sure what they are doing differently but balena etcher fails to even write the image do to some change in the compression they are using.

toastfr35, 2024-12-30:
The sha256sum I get on the .img.xz file after downloading does not match the on in the repo ...  damaged file?

Pneumanifest, 2024-12-30:
> The sha256sum I get on the .img.xz file after downloading does not match the one in the repo ... damaged file?

Unzip the file first.

toastfr35, 2024-12-30:
Thanks, I didn't spot they change the sha256 to be on the .img instead of the .xz has it was on 2.3.4

bjull001, 2025-01-02:
Hello again, 
Thank you for the answers. 
When checking the sha256 I get two different results, no matter if I check the img.xz or just the .img
Also, just trying to unpack the img. doesn't work either and gives me the same error as above. 
I also tried to compile my own image but got the same error even though the compiling was successful. 

Would love to know if this is a MacOS thing or if it persists on Windows or Linux. 
Thanks.

ecoinomist, 2025-01-13:
same issue here, I had the same problem 6 months ago when trying to install ArmBian, and gave up. Used the CB1 image without ArmBian. It must be something from ArmBian incompatible with MacOS

bjull001, 2025-01-20:
@ecoinomist Thank you for your answer. Could you describe how you used the CB1 image without armbian? What modifications have to be done? Thank you!

dahenson54, 2025-01-22:
I'm having the same issue with the CB1 V1.0 on a Mac Mini M2.

dahenson54, 2025-01-22:
Update: I was able to get mine to work. I unzipped the file and burned it to the card using Balena. I still got the message when it was complete, but I selected "Ignore" and ejected the card. It loaded into the CB1 and booted up without problems.

infinitejest, 2025-02-01:
The new image does not work. But if you use the previous version kernell 5.16 works. I guess we are stuck with old version until new one is updated.