bigtreetech/CB1 #199: CB1 + Manta M4P + V3.0.0  = TFT35 not work

artful172, 2025-01-03:
Hello
CB1 + Manta M4P + V3.0.0

at the moment of loading the image is there, and during work but instead of a KlipperScreen on the screen there is a black screen with a cursor
V2.3.4 works fine

config: 
[armbianEnv.txt](https://github.com/user-attachments/files/18300780/armbianEnv.txt)
[image]
[image]
[image]

shawnchr, 2025-01-03:
did update this section in the /boot/systemcfg
[image]

artful172, 2025-01-04:
me edit /boot/system.cfg , after me reboot, the problem remains

`###########################################
## klipperScreen Target Screen
## ks_src: "HDMI-1", "TFT35"
#ks_src="HDMI-1"
ks_src="TFT35"

## ks_angle: Rotation angle
##       normal: 0; inverted: 180;
##       left: 90; right: 270;
ks_angle="normal"

###########################################`

Zinput, 2025-01-21:
I have the same issue with Manta M8P + TFT35 SPI. You can use Ctrl + Alt + F1 or F3 to enter a different TTY and use the console, but it seems like the panfrost gpu driver isn't detecting the TFT35 SPI as a screen and only sees it as a console output.

artful172, 2025-02-08:
I couldn't, I installed the previous version

davey242, 2025-02-12:
I have Manta M8P with CB1 and TFT35 SPI 2.1
It started out white screen and then I uncommented overlays=tft35_spi
My armbianEnv.txt is now identical to file from original post.
Then I get the black screen with cursor like in original post. 

I then edited system.cfg to uncomment/change ks_src="TFT35" and after save/reboot the TFT35 is now working.

Coyoteman, 2025-03-17:
There is no /boot/BoardEnv.txt file in this version.  There is a similar file, armbianEnv.txt, which seems to be the same type of configurations, but the TFT35 does not work.

After creating a new boot SD card with the previous version, my display works.

juhjar, 2025-03-24:
Anyone got this combination to work? Black screen here too.

dmpl487, 2025-03-24:
> Anyone got this combination to work? Black screen here too.

Yes I will post my armbian and config when I get home

juhjar, 2025-03-25:
> > Anyone got this combination to work? Black screen here too.
> 
> Yes I will post my armbian and config when I get home

😉

dmpl487, 2025-03-25:
> > > Anyone got this combination to work? Black screen here too.
> > 
> > 
> > Yes I will post my armbian and config when I get home
> 
> 😉

[armbianEnv.txt](https://github.com/user-attachments/files/19458262/armbianEnv.txt)

[printer.txt](https://github.com/user-attachments/files/19458264/printer.txt)

so these are my config and Armbian, ymmv but this got me up and running with the BTT tft35 spi to an e3ez

StackingLayers, 2025-03-26:
overlays=tft35_spi

> Hello CB1 + Manta M4P + V3.0.0
>
> at the moment of loading the image is there, and during work but instead of a KlipperScreen on the screen there is a black screen with a cursor V2.3.4 works fine [...]

According to the images provided the screen is working fine. The curser only screen is signaling that klipperscreen isn't starting up correctly. Have you made sure that klipperscreen has the needed root rights? This command then reboot.
```
sudo bash -c "echo needs_root_rights=yes>>/etc/X11/Xwrapper.config"
```

crstian19, 2025-04-01:
I am using a generic, up-to-date Armbian installation on my BTT CB1. I have connected a BTT TFT35 SPI display. The display itself seems to initialize correctly, as it shows the TTY login console after booting.

However, KlipperScreen does not appear on the TFT35 SPI display; only the TTY console remains visible. The KlipperScreen service appears to be running when checked via systemctl, but it's not rendering its GUI on the SPI display. 

What I've 

- Ensured KlipperScreen is installed and the service is enabled/running.
- Attempted to load Device Tree Overlays suspected to be for this display (like tft35_spi, potentially sourced from BTT's repositories) via /boot/armbianEnv.
- Verified that loading the overlay often results in a /dev/fb1 device being created.

StackingLayers, 2025-04-02:
> However, KlipperScreen does not appear on the TFT35 SPI display; only the TTY console remains visible.

Have you tried what I posted in the reply above? Klipperscreen needs root rights.

juhjar, 2025-04-02:
> I am using a generic, up-to-date Armbian installation on my BTT CB1. I have connected a BTT TFT35 SPI display. The display itself seems to initialize correctly, as it shows the TTY login console [...]

Which board you have?

EidenPride, 2025-05-26:
Hi, did you solve this problem, login tty1 screen here too.

p4block, 2025-05-29:
Works on my machine after uncommenting the two lines in the /boot/system.cfg. Now, the rotation part (180) seems to only apply to the visuals, as the input is still treated as not rotated. This means I need to touch on the opposite corner of where a button is 😬

p4block, 2025-05-29:
Read KlipperScreen docs and apparently this needs to be fixed either in libinput or rebuilding the firmware of the screen

First option was quicker, my screen is flipped 180 degrees. The trick about using udev rules to inject an env var into libinput didn't work for some reason.I had to resort to the "fallback method" of just slapping a xorg config. 

```
biqu@bigtreetech-cb1:~$ cat /usr/share/X11/xorg.conf.d/99-rotate-touchscreen.conf
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option "TransformationMatrix" "-1 0 1 0 -1 1 0 0 1"
EndSection
```