bigtreetech/CB1 #202: BTT Image 3.0 Error imaging on mac

infinitejest, 2025-02-01:
The image for CB2_Debian_Klipper_Kernel6.6_2024 is causing issues when you image into the SD card. Mac Os is not able to recognize it. I try to image the kernell 5.16 and the card was imaged and sucessful recognize in the mac os. 

OS: MacOSX 13.71
Imager: Raspberry Imager
SD Card: SanDisk high endurance 64 GB.

danieleotto, 2025-03-03:
Same problem here, seems my sha256 is not valid for the downloaded file

WorldsWorstAstronaut, 2025-03-12:
Same here. It almost looks like the image isn't partitioning the card correctly?

ismirv, 2025-03-20:
Same situation. CB1_Debian_Klipper_Kernel6.6_2024 after flashing the SD card is not readable on either Mac or Windows

thehamstermuffin, 2025-04-12:
Same here kernel 6.6 is not working

Pneumanifest, 2025-04-13:
I don't know how it works with Mac, but with Windows you will not be able to read the Linux OS that is flashed to the SD card. It's a completely different operating system, and a file format not used by Windows. Windows will throw a corrupted drive error message because of this. You must ignore those. As long as you ignore the, 'need to format' messages The only partition that should be readable is the boot partition.

Talha909, 2025-08-28:
Same problem here. What should i do

Talha909, 2025-08-28:
> I have the same issue. After flashing the image CB01 V 3.0.0 on my Mac. The system does not recognize the SD Cart. It is impossible to configure WiFi setting. Could you tel me what to do ?

Did you find anything.