bigtreetech/CB1 #203: Complete system freeze during print, happening randomly

ropa26, 2025-02-08:
Hi,

I have the CB1 on a Manta M8P board. Sometimes I can print perfectly fine, but sometimes during longer prints the printer stops and the system completely freezes. I cannot use the display, and network connectivity is also gone.

First I thought it is due to high temperatures, so I added a cooling fan to the board. CB1 board temperature during print is 58°C.
Then I thought it is due to WiFi driver issues, so I disabled WiFi and instead used the LAN connection. But the freeze still happens.

In the logs (syslog, kern.log) I don't see any messages from before the freeze, so I cannot debug the problem further.

I would appreciate any help or ideas on how to solve it or where the issue could be.

Thanks!

# System information:
- Voron 2.4 printer
- CB1 firmware version:  v3.0.0-25.02.0-trunk rolling
- Kernel: 6.6.66-vendor-sunxi64

JRMSBr, 2025-02-09:
I installed version 3 minimal and it freezes when I try to install kiauh

lupustr3, 2025-02-15:
Same thing here. Random system hangs on 3.0.0 image. Using together with Ender 3 v3 se. It can work without issues for many hours and then just becomes unresponsive in any way.

lupustr3, 2025-02-15:
journal of hanged boot, no entries on hang time (~16-17 hours)
```
biqu@bigtreetech-cb1:~$ journalctl -p 4 -b -1
Feb 15 15:17:04 bigtreetech-cb1 kernel: KASLR disabled due to lack of seed
Feb 15 15:17:04 bigtreetech-cb1 kernel: cacheinfo: Unable to detect cache hierarchy for CPU 0
Feb 15 15:17:04 bigtreetech-cb1 kernel: gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
Feb 15 15:17:04 bigtreetech-cb1 kernel: gpio gpiochip1: Static allocation of GPIO base is deprecated, use dynamic allocation.
Feb 15 15:17:04 bigtreetech-cb1 kernel: debugfs: Directory '1100000.mixer' with parent 'regmap' already present!
Feb 15 15:17:04 bigtreetech-cb1 kernel: debugfs: Directory '1100000.mixer' with parent 'regmap' already present!
Feb 15 15:17:04 bigtreetech-cb1 kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-ph not found, using dummy regulator
Feb 15 15:17:04 bigtreetech-cb1 kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pc not found, using dummy regulator
Feb 15 15:17:04 bigtreetech-cb1 kernel: usb_phy_generic usb_phy_generic.3.auto: dummy supplies not allowed for exclusive requests
Feb 15 15:17:04 bigtreetech-cb1 kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pa not found, using dummy regulator
Feb 15 15:17:05 bigtreetech-cb1 kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pg not found, using dummy regulator
Feb 15 15:17:05 bigtreetech-cb1 kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pf not found, using dummy regulator
Feb 15 15:17:05 bigtreetech-cb1 kernel: mmc0: host does not support reading read-only switch, assuming write-enable
Feb 15 15:17:05 bigtreetech-cb1 kernel: sunxi-mmc 4021000.mmc: card claims to support voltages below defined range
Feb 15 15:17:05 bigtreetech-cb1 kernel: device-mapper: core: CONFIG_IMA_DISABLE_HTABLE is disabled. Duplicate IMA measurements will not be recorded in the IMA log.
Feb 15 15:17:05 bigtreetech-cb1 kernel: it's ac300, skip the ac200 init!
Feb 15 15:17:05 bigtreetech-cb1 kernel: ac200-sunxi: probe of 2-0010 failed with error -22
Feb 15 15:17:05 bigtreetech-cb1 systemd-tmpfiles[530]: Setting file flags is only supported on regular files and directories, cannot set on '/var/log/journal'.
Feb 15 15:17:06 bigtreetech-cb1 avahi-daemon[595]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
```

lupustr3, 2025-03-02:
Since previous post I switched to 2.3.4 (using same version of klipper) and no hangs ever occurred during this period

crysxd, 2025-03-07:
I can confirm the same issue on my CB1, but I for sure do not have the 3.0.0. Not sure where I can check what version I have, but I flashed the SD in summer...so most likely `2.3.4`

The machine freezes and is completely unresponsive to any network requests

fulafisken, 2025-03-26:
I have a simular issure. These are some outputs from my syslog. It seems to be happening randomly when the CB1 is used, sometimes it can work for an hour or two, and sometimes it crasches within minutes from boot. I am going to try to downgrade to 2.3.4 to see if that solves it.

```
Message from syslogd@voron24 at Mar 26 06:15:50 ...
 kernel:[  156.145461] Internal error: Oops: 0000000096000004 [#1] SMP

Message from syslogd@voron24 at Mar 26 06:15:50 ...
 kernel:[  156.340772] Code: d1000463 f9000a43 34001220 51000403 (f863d843) 
Mar 26 06:15:50 voron24 kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
Mar 26 06:15:50 voron24 kernel: Mem abort info:
Mar 26 06:15:50 voron24 kernel:   ESR = 0x0000000096000004
Mar 26 06:15:50 voron24 kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Mar 26 06:15:50 voron24 kernel:   SET = 0, FnV = 0
Mar 26 06:15:50 voron24 kernel:   EA = 0, S1PTW = 0
Mar 26 06:15:50 voron24 kernel:   FSC = 0x04: level 0 translation fault
Mar 26 06:15:50 voron24 kernel: Data abort info:
Mar 26 06:15:50 voron24 kernel:   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
Mar 26 06:15:50 voron24 kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Mar 26 06:15:50 voron24 kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Mar 26 06:15:50 voron24 kernel: user pgtable: 4k pages, 48-bit VAs, pgdp=0000000046cbb000
Mar 26 06:15:50 voron24 kernel: [0000000000000000] pgd=0000000000000000, p4d=0000000000000000
Mar 26 06:15:50 voron24 kernel: Internal error: Oops: 0000000096000004 [#1] SMP
Mar 26 06:15:50 voron24 kernel: Modules linked in: can_raw can sunrpc ac200_phy lz4hc lz4 zram binfmt_misc nls_iso8859_1 polyval_ce polyval_generic 8189fs gs_usb snd_soc_sunxi_machine snd_soc_sunxi_ahub can_dev snd_soc_hdmi_codec sun50iw9_codec snd_soc_sunxi_ahub_dam cfg80211 rfkill panfrost gpu_sched pwm_sunxi_enhance drm_shmem_helper joydev cpufreq_dt fuse dm_mod hid_multitouch dw_hdmi_cec dw_hdmi_i2s_audio sunxi_gmac sunxi_ephy sunxi_ac200 display_connector
Mar 26 06:15:50 voron24 kernel: CPU: 0 PID: 1231 Comm: Xorg Not tainted 6.6.66-vendor-sunxi64 #4
Mar 26 06:15:50 voron24 kernel: Hardware name: BigTreeTech CB1 (DT)
Mar 26 06:15:50 voron24 kernel: pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Mar 26 06:15:50 voron24 kernel: pc : mas_prev_slot+0xd8/0x634
Mar 26 06:15:50 voron24 kernel: lr : mas_prev_range+0xb8/0x128
Mar 26 06:15:50 voron24 kernel: sp : ffff80008338bbb0
Mar 26 06:15:50 voron24 kernel: x29: ffff80008338bbb0 x28: 0000000000000000 x27: 0000ffffb0434000
Mar 26 06:15:50 voron24 kernel: x26: 0000ffffb0431000 x25: 0000000000000001 x24: ffff0000030e97c0
Mar 26 06:15:50 voron24 kernel: x23: 0000000000000002 x22: 0000000000000001 x21: 0000000000000001
Mar 26 06:15:50 voron24 kernel: x20: 0000000000000009 x19: 0000000000000000 x18: ffff80008338bdb8
Mar 26 06:15:50 voron24 kernel: x17: 0000000000000000 x16: ffff800080f6c490 x15: 0000ffffb04f6fff
Mar 26 06:15:50 voron24 kernel: x14: ffff800080f6c490 x13: 0000ffffb04cffff x12: 0000ffffb04c0fff
Mar 26 06:15:50 voron24 kernel: x11: 0000ffffb04bffff x10: 0000000000000006 x9 : 0000000000000001
Mar 26 06:15:50 voron24 kernel: x8 : ffff00003a2ae000 x7 : 0000000000000004 x6 : 0000000000000000
Mar 26 06:15:50 voron24 kernel: x5 : 0000000000000001 x4 : ffff0000046e991c x3 : 0000000000000000
Mar 26 06:15:50 voron24 kernel: x2 : 0000000000000000 x1 : ffff000005c10d00 x0 : 0000000000000001
Mar 26 06:15:50 voron24 kernel: Call trace:
Mar 26 06:15:50 voron24 kernel:  mas_prev_slot+0xd8/0x634
Mar 26 06:15:50 voron24 kernel:  mas_prev_range+0xb8/0x128
Mar 26 06:15:50 voron24 kernel:  do_vmi_align_munmap.constprop.0+0x270/0x424
Mar 26 06:15:50 voron24 kernel:  do_vmi_munmap+0xc4/0x118
Mar 26 06:15:50 voron24 kernel:  __vm_munmap+0x94/0x110
Mar 26 06:15:50 voron24 kernel:  __arm64_sys_munmap+0x20/0x30
Mar 26 06:15:50 voron24 kernel:  invoke_syscall+0x48/0x118
Mar 26 06:15:50 voron24 kernel:  el0_svc_common.constprop.0+0x40/0xe8
Mar 26 06:15:50 voron24 kernel:  do_el0_svc+0x20/0x2c
Mar 26 06:15:50 voron24 kernel:  el0_svc+0x38/0xd8
Mar 26 06:15:50 voron24 kernel:  el0t_64_sync_handler+0x13c/0x158
Mar 26 06:15:50 voron24 kernel:  el0t_64_sync+0x190/0x194
Mar 26 06:15:50 voron24 kernel: Code: d1000463 f9000a43 34001220 51000403 (f863d843) 
Mar 26 06:15:50 voron24 kernel: ---[ end trace 0000000000000000 ]---

```

danwig, 2025-05-03:
BTT??? Are you there?

omidta, 2025-05-21:
Same thing here:

```
biqu@Smaukh:~$ journalctl -p 4 -b -1
May 22 08:17:05 Smaukh kernel: KASLR disabled due to lack of seed
May 22 08:17:05 Smaukh kernel: cacheinfo: Unable to detect cache hierarchy for CPU 0
May 22 08:17:05 Smaukh kernel: gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
May 22 08:17:05 Smaukh kernel: gpio gpiochip1: Static allocation of GPIO base is deprecated, use dynamic allocation.
May 22 08:17:05 Smaukh kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-ph not found, using dummy regulator
May 22 08:17:05 Smaukh kernel: debugfs: Directory '1100000.mixer' with parent 'regmap' already present!
May 22 08:17:05 Smaukh kernel: debugfs: Directory '1100000.mixer' with parent 'regmap' already present!
May 22 08:17:05 Smaukh kernel: usb_phy_generic usb_phy_generic.3.auto: dummy supplies not allowed for exclusive requests
May 22 08:17:05 Smaukh kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pa not found, using dummy regulator
May 22 08:17:05 Smaukh kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pg not found, using dummy regulator
May 22 08:17:05 Smaukh kernel: sun50i-h616-pinctrl 300b000.pinctrl: supply vcc-pf not found, using dummy regulator
May 22 08:17:05 Smaukh kernel: mmc0: host does not support reading read-only switch, assuming write-enable
May 22 08:17:05 Smaukh kernel: sunxi-mmc 4021000.mmc: card claims to support voltages below defined range
May 22 08:17:05 Smaukh kernel: device-mapper: core: CONFIG_IMA_DISABLE_HTABLE is disabled. Duplicate IMA measurements will not>
May 22 08:17:05 Smaukh kernel: it's ac300, skip the ac200 init!
May 22 08:17:05 Smaukh kernel: ac200-sunxi: probe of 2-0010 failed with error -22
May 22 08:17:05 Smaukh systemd-journald[533]: File /var/log/journal/492fc8281052434a8b3b5a24b92c3762/system.journal corrupted >
May 22 08:17:05 Smaukh systemd-tmpfiles[536]: Setting file flags is only supported on regular files and directories, cannot se>
May 22 08:17:06 Smaukh avahi-daemon[600]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
lines 1-19/19 (END)
```