bigtreetech/CB1 #204: BTT Pi 1.2 image V2.3.4 - root password?

UndeadBrx, 2025-02-14:
Hello, what is the password for the "root" user in this image? 'biqu', 'root', 'toor', 'pass' do not work.

UndeadBrx, 2025-02-16:
Problem solved.

The image has blocked `root` access via SSH.

You need to log in as `biqu` and unblock access.
Open the SSH configuration file: `sudo nano /etc/ssh/sshd_config`
Find the line: `PermitRootLogin no`
and change it to: `PermitRootLogin yes`
Save and exit: In the nano editor press: `CTRL+X → Y → Enter`
Reboot: `reboot`

Log in via SSH as `root` user: `root/root`

closed 2025-02-16 by UndeadBrx

Pneumanifest, 2025-02-16:
> Incorrect. Any other ideas?

Not incorrect. Default armbian root password is 1234 BTT has it set to root.

shawnchr, 2025-02-16:
Well 1234 BTT does not work without changing the permitrootlogin so not sure how your claiming incorrect.

Pneumanifest, 2025-02-16:
The question was, " root password?" The password, "root" was given and "1234" as an alternative option since that is the default armbian root password at a fresh un-modified install.
I see they figured out the real problem. The question said nothing about how to gain SSH login permission. That has nothing to do with the root password. And what do you know, the password was root. Nothing was incorrect about the other reply.