bigtreetech/CB1 #205: Image SHA256 checksum mismatch

redhavoc, 2025-03-07:
I get a diiferent hash key between the CB1 Klipper image and the provided sha256 key. I tried to download the image 3 times to make sure but they all seem to provide the same incorrect hashkey. Also this image doesn't seem to work in my sd card so probably something is wrong

#~/Downloads$ sha256sum CB1_Debian12_Klipper_kernel6.6_20241219.img.xz 
eff1412d4525c1ce9c6fe7135a771b2c507bfda5caf595a2b22735b68ebb9b24  CB1_Debian12_Klipper_kernel6.6_20241219.img.xz

#~/Downloads$ more CB1_Debian12_Klipper_kernel6.6_20241219.img.sha256 
f9d9143bcb47ad6bd859c24eb9f06093385b9957d89e6493e520c401150584d7

can someone take a look at it please and verify.

redhavoc, 2025-03-08:
I think its a duplicate of [#202 ](https://github.com/bigtreetech/CB1/issues/202)

valir, 2025-05-18:
I've also noticed this difference, but then I've however decompressed the xz file and found out that the extracted img's sha256sum was matching.