bigtreetech/CB1 #206: Current release does not flash to sd using balena etcher and does not have correct sha256

tjleon1, 2025-04-04:
### The current releases have the following issues:
<p>The Signature files do not match the downloaded image files and the image files fail to flash using Balena Etcher. Using sha256sum on the following files yields the sum listed below it, but the sum you provided (below that) as you can see do not match:</p>
<p>CB1_Debian12_Klipper_kernel6.6_20241219.img.xz</p>
<table style="border-collapse: collapse; width: 100%; height: 36px;" border="1">
<tbody>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">Generated sha256sum</td>
<td style="width: 50%; height: 18px;"><span style="color: #ff0000;"><code>eff1412d4525c1ce9c6fe7135a771b2c507bfda5caf595a2b22735b68ebb9b24</code></span></td>
</tr>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">CB1_Debian12_Klipper_kernel6.6_20241219.img.sha256</td>
<td style="width: 50%; height: 18px;"><code>f9d9143bcb47ad6bd859c24eb9f06093385b9957d89e6493e520c401150584d7</code></td>
</tr>
</tbody>
</table>
<p>CB1_Debian12_minimal_kernel6.6_20241219.img.xz</p>
<table style="border-collapse: collapse; width: 100%; height: 54px;" border="1">
<tbody>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">Generated sha256sum</td>
<td style="width: 50%; height: 18px;"><span style="color: #ff0000;"><code>b60f863f3a98d5e23642635aeca8d4f96550ddb470200aa8f14506f59d83829c</code></span></td>
</tr>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">CB1_Debian12_minimal_kernel6.6_20241219.img.sha256</td>
<td style="width: 50%; height: 18px;"><code>3d29eb529ab76b28190b73a5aa32e3864e1e9ca1229f41f8035da52f0c892b10</code></td>
</tr>
</tbody>
</table>
<p>**What I discovered:**</p>
<p>Normally, one would take the sha256 from the actual file being offered for download, but after looking closely at the naming of the files I decided to extract the files from xz format since the sha256 filenames did not have .xz in them. So if I extract the file so that I truly am taking the sha256 of the image file as the name of the sha256 file would seem to indicate I get:</p>
<p>CB1_Debian12_Klipper_kernel6.6_20241219.img (without the .xz)</p>
<table style="border-collapse: collapse; width: 100%; height: 36px;" border="1">
<tbody>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">Generated sha256sum</td>
<td style="width: 50%; height: 18px;"><span><code style="color: #008000;">f9d9143bcb47ad6bd859c24eb9f06093385b9957d89e6493e520c401150584d7</code></span></td>
</tr>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">CB1_Debian12_Klipper_kernel6.6_20241219.img.sha256</td>
<td style="width: 50%; height: 18px;"><code>f9d9143bcb47ad6bd859c24eb9f06093385b9957d89e6493e520c401150584d7</code></td>
</tr>
</tbody>
</table>
<p>CB1_Debian12_minimal_kernel6.6_20241219.img (without the .xz)</p>
<table style="border-collapse: collapse; width: 100%; height: 54px;" border="1">
<tbody>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">Generated sha256sum</td>
<td style="width: 50%; height: 18px;"><span><code style="color: #008000;">3d29eb529ab76b28190b73a5aa32e3864e1e9ca1229f41f8035da52f0c892b10</code></span></td>
</tr>
<tr style="height: 18px;">
<td style="width: 50%; height: 18px;">CB1_Debian12_minimal_kernel6.6_20241219.img.sha256</td>
<td style="width: 50%; height: 18px;"><code>3d29eb529ab76b28190b73a5aa32e3864e1e9ca1229f41f8035da52f0c892b10</code></td>
</tr>
</tbody>
</table>
<p>Since the download doesn't match the provided signature, it tends to make potential downloaders doubt the veracity of the download and potentially not use it. Furthermore, there is no instruction to extract the file prior to use, which would be very helpful. This is very confusing to encounter if you don't know to look VERY closely at the filenames.</p>
<p>Before coming onto the idea to extract first, I tried to use the previous release, and I had no problems with verifying since the sha256 was taken from the compressed file as I here suggest. I was also able to flash that older .xz version to an SD card without extracting it using Balena Etcher - I did not however attempt to use that before the idea of extracting first occurred to me. </p>
<p>Since it was able to flash the older .xz version, I'm not sure that it would have worked or if it had, why the newer .xz file would not flash at all.</p>
<p>My suggested change would be to provide the sha256 for the compressed file users will download, and then offer an instruction to extract the image file before using it.</p>

Pneumanifest, 2025-04-04:
You should still extract first when using balena etcher. There's an odd issue where you can still cause errors and failures during check. Or use Raspberry Pi imager as it automatically pre extracts correctly before trying to flash.

tjleon1, 2025-04-16:
> You should still extract first when using balena etcher. There's an odd known issue where it can still cause errors and failures during check. Or use Raspberry Pi imager as it automatically pre extracts correctly before trying to flash.

I agree that you should extract before burning it with Balena Etcher, but that isn't obvious from  the readme.

There is still the issue with the wrong SHA256 sum provided. Common practice is that the provided SHA256 sum is for whatever you download, not for what it contains after completing another step (extracting).