bigtreetech/CB1 #220: CB1 disconnection issues

Sangra6969, 2025-09-04:
Do they still update this firmware? I always seem to get disconnection issues with the V3.0 release. I went back to the V2.3.4 update and so far within 2hrs no disconnection issues. Someone screwed up somewhere in the programming.

larsboelling, 2025-09-07:
> Do they still update this firmware? I always seem to get disconnection issues with the V3.0 release. I went back to the V2.3.4 update and so far within 2hrs no disconnection issues. Someone screwed up somewhere in the programming.

+1

Sangra6969, 2025-10-18:
> I've seen tons of disconnection issues; but I'm seeing them in 2.3.4 as well. May be other things going on with this device.

I haven't had any issues so far with 2.3.4